amaranth-lang/amaranth #548: Unusual code in Operator for << and >>

RobertBaruch, 2020-11-21:
This didn't make sense to me:

```python
>>> from nmigen import *
>>> s1 = Signal(4)
>>> s1.shape()
unsigned(4)
>>> s1.shift_left(2).shape()
unsigned(6)
>>> (s1 << 2).shape()
unsigned(7)
```

The code causing this comes from `Operator`:

```python
            if self.operator == "<<":
                if b_signed:
                    extra = 2 ** (b_width - 1) - 1
                else:
                    extra = 2 ** (b_width)     - 1
                return Shape(a_width + extra, a_signed)
            if self.operator == ">>":
                if b_signed:
                    extra = 2 ** (b_width - 1)
                else:
                    extra = 0
                return Shape(a_width + extra, a_signed)
```

Effectively, the constness of the shift amount is not used.

Two points:

-  This works by taking into account shifting by a non-const signal. 
-  `__check_shamt` prevents shifting by a signed `Value`, so `b_signed` will never be true.

Is it okay to put some extra code in there to handle the case where we are shifting by a `Const` and be a bit more surgical in the computing the new width?

whitequark, 2020-11-21:
> Effectively, the constness of the shift amount is not used.

This is intentional. Basically, if we special-case `x << 1`, then should we special-case `x << (1 + 1)`? "No constant folding is ever performed when determining the shape of a value" is a nice simple answer. Constant folding is not.

> * `__check_shamt` prevents shifting by a signed `Value`, so `b_signed` will never be true.

That's a leftover of some very early changes. We can get rid of that.

closed 2020-11-21 by RobertBaruch

whitequark, 2020-11-21:
I'm curious, do you have a case where constant folding would help you? In general shape determination in nMigen is very conservative, so I'm not sure if fixing this particular instance would help a lot.

RobertBaruch, 2020-11-21:
No, I just thought it was a bit odd to see that. Normally truncation "just works":

```python
x = Signal(32)
y = Signal(32)

m.d.comb += y.eq(x << 5)
m.d.comb += Assert(y == (x << 5)[:32])
```

so it never really matters.