amaranth-lang/amaranth #550: Attrs on DiffPairs only generate constraints for positive side

anuejn, 2020-11-22:
When I apply an io constraint to a `DiffPair` on machxo2 via `Attrs` the constraint only gets applied to the positive side of the differential pair. While this is fine in most cases for the `DIFFRESISTOR=100` attribute this leads to only one side being terminated and thus SI problems.

Repro:
```python
from nmigen import *
from nmigen.build import *
from nmigen_boards.tinyfpga_ax2 import *

class Test(Elaboratable):
    def elaborate(self, plat):
        m = Module()

        plat.add_resources([Resource("diff_pair", 0, DiffPairs("1", "2", conn=("gpio", 0), dir="i"), Attrs(IO_TYPE="LVDS25", DIFFRESISTOR="100"))])
        plat.add_resources([Resource("gpio", 0, Pins("3", conn=("gpio", 0), dir="o"), Attrs(IOSTANDARD="LVCMOS33"))])

        m.d.comb += plat.request("gpio", 0).eq(plat.request("diff_pair", 0))

        return m

TinyFPGAAX2Platform().build(Test())
```

leads to the following lpf:
```diff
# Automatically generated by nMigen 0.3.dev225+g56b899e.dirty. Do not edit.
BLOCK ASYNCPATHS;
BLOCK RESETPATHS;
LOCATE COMP "gpio_0__io" SITE "16";
IOBUF PORT "gpio_0__io" IOSTANDARD=LVCMOS33;
LOCATE COMP "diff_pair_0__p" SITE "13";
IOBUF PORT "diff_pair_0__p" IO_TYPE=LVDS25 DIFFRESISTOR=100;
# (add_preferences placeholder)
```

whitequark, 2020-11-22:
> While this is fine in most cases

This is not just *fine* in most cases, it is in fact a *hard requirement*.

> While this is fine in most cases for the `DIFFRESISTOR=100` attribute this leads to only one side being terminated and thus SI problems.

This means it is necessary to redesign the code that constrains differential ports. Please research which attributes need to be applied to one pin of the differential pair, and which attributes need to be applied to both. After that we can proceed.

anuejn, 2020-11-22:
Hm... I was unable to find any information in the lattice docs about this :(.
However, while fiddeling around with diamond setting IO_TYPE  and SITE on both p/n worked.

I dont really know how to proceed here

whitequark, 2020-11-22:
Maybe @daveshah1 has an idea?

daveshah1, 2020-11-22:
This is very strange, for ECP5 physically there is only one bitstream bit that enables the diffresistor for both A and B

whitequark, 2020-11-22:
@anuejn Can you diff the bitstreams with DIFFRESISTOR on none, one, two pins of the pair?

daveshah1, 2020-11-22:
It's even more interesting that [Lattice's own example](https://www.latticesemi.com/-/media/LatticeSemi/Documents/ReferenceDesigns/MIPI/CSI2RX/CSI2RXMachXO28bit4laneRAW8.ashx) also only puts the constraint on the positive side:

```
LOCATE COMP "rstn" SITE "C1" ;
LOCATE COMP "DCK" SITE "N6" ;
LOCATE COMP "CH0" SITE "M11" ;
LOCATE COMP "CH1" SITE "P8" ;
LOCATE COMP "CH2" SITE "P2" ;
LOCATE COMP "CH3" SITE "M7" ;
LOCATE COMP "sensor_clk" SITE "C8" ;
LOCATE COMP "pixclk_adj" SITE "A11" ;
LOCATE COMP "fv" SITE "B7" ;
LOCATE COMP "lv" SITE "C4" ;
LOCATE COMP "pixdata_0" SITE "C6" ;
LOCATE COMP "pixdata_1" SITE "B3" ;
LOCATE COMP "pixdata_2" SITE "C11" ;
LOCATE COMP "pixdata_3" SITE "A12" ;
LOCATE COMP "pixdata_4" SITE "A7" ;
LOCATE COMP "pixdata_5" SITE "B5" ;
LOCATE COMP "pixdata_6" SITE "A9" ;
LOCATE COMP "pixdata_7" SITE "A10" ;
LOCATE COMP "pixdata_8" SITE "A2" ;
LOCATE COMP "pixdata_9" SITE "B12" ;
IOBUF PORT "CH0" IO_TYPE=LVDS25 DIFFRESISTOR=100 ;
IOBUF PORT "CH1" IO_TYPE=LVDS25 DIFFRESISTOR=100 ;
IOBUF PORT "CH2" IO_TYPE=LVDS25 DIFFRESISTOR=100 ;
IOBUF PORT "CH3" IO_TYPE=LVDS25 DIFFRESISTOR=100 ;
IOBUF PORT "DCK" IO_TYPE=LVDS25 DIFFRESISTOR=100 ;
FREQUENCY NET "pixclk" 148.500000 MHz ;
FREQUENCY NET "pixclk_adj_c" 148.500000 MHz ;
FREQUENCY NET "sclk_in_c" 37.125000 MHz ;
FREQUENCY NET "deser/eclko" 185.625000 MHz ;
```

anuejn, 2020-11-22:
There seems to be a difference?

``` diff
[anuejn@pink build]$ diff none.bit.hex one.bit.hex 
12c12
< 000000b0: 204e 6f76 2032 3220 3136 3a30 333a 3338   Nov 22 16:03:38
---
> 000000b0: 204e 6f76 2032 3220 3136 3a30 343a 3438   Nov 22 16:04:48
18c18
< 00000110: 5243 3a20 3078 3035 3738 0043 5243 2043  RC: 0x0578.CRC C
---
> 00000110: 5243 3a20 3078 4431 3333 0043 5243 2043  RC: 0xD133.CRC C
367,369c367,369
< 000016e0: 0000 0000 0000 0000 0000 0000 0000 0001  ................
< 000016f0: 1cf3 ca00 0000 0000 0000 0000 0000 0000  ................
< 00001700: 0000 0001 1cf5 0500 0000 0000 0000 0000  ................
---
> 000016e0: 0000 0000 0000 0000 0000 0000 0000 0004  ................
> 000016f0: 28e7 9e50 0000 0000 0000 0000 0000 0000  (..P............
> 00001700: 0000 0000 0428 e7a8 2800 0000 0000 0000  .....(..(.......
376,381c376,381
< 00001770: 9400 0560 0000 0000 04e0 0009 c000 1380  ...`............
< 00001780: 0000 9c00 0138 0002 7000 04e0 0000 0000  .....8..p.......
< 00001790: 0000 0000 0000 0000 0000 0000 0000 55b7  ..............U.
< 000017a0: ffff ffff ffff ffff c280 0000 0000 0000  ................
< 000017b0: 2aa7 2200 0000 4000 0000 5e00 0000 ffff  *."...@...^.....
< 000017c0: ffff                                     ..
---
> 00001770: 0000 9400 0560 0000 0000 04e0 0009 c000  .....`..........
> 00001780: 1380 0000 9c00 0138 0002 7000 04e0 0000  .......8..p.....
> 00001790: 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 000017a0: f1a7 ffff ffff ffff ffff c280 0000 0000  ................
> 000017b0: 0000 2aa7 2200 0000 4000 0000 5e00 0000  ..*."...@...^...
> 000017c0: ffff ffff                                ....
[anuejn@pink build]$ diff two.bit.hex one.bit.hex 
12c12
< 000000b0: 204e 6f76 2032 3220 3136 3a30 353a 3535   Nov 22 16:05:55
---
> 000000b0: 204e 6f76 2032 3220 3136 3a30 343a 3438   Nov 22 16:04:48
[anuejn@pink build]$ diff none.bit.hex two.bit.hex 
12c12
< 000000b0: 204e 6f76 2032 3220 3136 3a30 333a 3338   Nov 22 16:03:38
---
> 000000b0: 204e 6f76 2032 3220 3136 3a30 353a 3535   Nov 22 16:05:55
18c18
< 00000110: 5243 3a20 3078 3035 3738 0043 5243 2043  RC: 0x0578.CRC C
---
> 00000110: 5243 3a20 3078 4431 3333 0043 5243 2043  RC: 0xD133.CRC C
367,369c367,369
< 000016e0: 0000 0000 0000 0000 0000 0000 0000 0001  ................
< 000016f0: 1cf3 ca00 0000 0000 0000 0000 0000 0000  ................
< 00001700: 0000 0001 1cf5 0500 0000 0000 0000 0000  ................
---
> 000016e0: 0000 0000 0000 0000 0000 0000 0000 0004  ................
> 000016f0: 28e7 9e50 0000 0000 0000 0000 0000 0000  (..P............
> 00001700: 0000 0000 0428 e7a8 2800 0000 0000 0000  .....(..(.......
376,381c376,381
< 00001770: 9400 0560 0000 0000 04e0 0009 c000 1380  ...`............
< 00001780: 0000 9c00 0138 0002 7000 04e0 0000 0000  .....8..p.......
< 00001790: 0000 0000 0000 0000 0000 0000 0000 55b7  ..............U.
< 000017a0: ffff ffff ffff ffff c280 0000 0000 0000  ................
< 000017b0: 2aa7 2200 0000 4000 0000 5e00 0000 ffff  *."...@...^.....
< 000017c0: ffff                                     ..
---
> 00001770: 0000 9400 0560 0000 0000 04e0 0009 c000  .....`..........
> 00001780: 1380 0000 9c00 0138 0002 7000 04e0 0000  .......8..p.....
> 00001790: 0000 0000 0000 0000 0000 0000 0000 0000  ................
> 000017a0: f1a7 ffff ffff ffff ffff c280 0000 0000  ................
> 000017b0: 0000 2aa7 2200 0000 4000 0000 5e00 0000  ..*."...@...^...
> 000017c0: ffff ffff                                ....
```

[bitstreams.zip](https://github.com/nmigen/nmigen/files/5579719/bitstreams.zip)

anuejn, 2020-11-22:
Oh no, there seems to be no difference! sorry for bothering you, I was simply wrong

closed 2020-11-22 by anuejn