amaranth-lang/amaranth #553: Failing Multidomain in CXXRTL

BracketMaster, 2020-12-02:
Started messing around with CXXRTL today and 
could not seem to get yosys to finish emitting CXXRTL.
I was trying to emit CXXRTL from a multi-domain nMIgen
as found here:

https://github.com/BracketMaster/nmigen-by-example/blob/master/cxxtrl-multi-domain.py

Using ``Yosys 0.9+3715 (git sha1 2945e270, clang 12.0.0 -fPIC -Os)``

I got the following error:
```bash
cxxtrl-multi-domain.py:99: DeprecationWarning: instead of nmigen.back.pysim.*, use nmigen.sim.*
  from nmigen.back.pysim import Simulator, Tick
Traceback (most recent call last):
  File "cxxtrl-multi-domain.py", line 104, in <module>
    f.write(cxxrtl.convert(top, ports=top.ports()))
  File "/Users/yehowshuaimmanuel/.virtualenvs/maeri38/lib/python3.8/site-packages/nmigen/back/cxxrtl.py", line 41, in convert
    return _convert_rtlil_text(rtlil_text, black_boxes, src_loc_at=1)
  File "/Users/yehowshuaimmanuel/.virtualenvs/maeri38/lib/python3.8/site-packages/nmigen/back/cxxrtl.py", line 31, in _convert_rtlil_text
    return yosys.run(["-q", "-"], "\n".join(script), src_loc_at=1 + src_loc_at)
  File "/Users/yehowshuaimmanuel/.virtualenvs/maeri38/lib/python3.8/site-packages/nmigen/_toolchain/yosys.py", line 186, in run
    return cls._process_result(popen.returncode, stdout, stderr, ignore_warnings, src_loc_at)
  File "/Users/yehowshuaimmanuel/.virtualenvs/maeri38/lib/python3.8/site-packages/nmigen/_toolchain/yosys.py", line 108, in _process_result
    raise YosysError(stderr.strip())
nmigen._toolchain.yosys.YosysError: ERROR: Assert `signal.is_wire() && signal.is_bit()' failed in backends/cxxrtl/cxxrtl_backend.cc:2024.
```

BracketMaster, 2020-12-02:
To be quite honest, I'm not sure this is strictly a multi-domain issue.

whitequark, 2020-12-02:
This happens when you have a clock input floating or connected to `1'bx` in the netlist.

BracketMaster, 2020-12-02:
Ah. I guess I should be asking what is the proper way to do multi domain
from nMigen to CXXRTL.

whitequark, 2020-12-02:
The crash will be fixed by https://github.com/YosysHQ/yosys/pull/2469.

whitequark, 2020-12-02:
> Ah. I guess I should be asking what is the proper way to do multi domain from nMigen to CXXRTL.

Regarding your code example: you are creating the `producer` and `consumer` domains, but never drive their clocks, so they are bound to constant 0 by default. You can add their clocks to ports, or just don't create the domains, in which case they'll be created automatically.

(This is a difference from how pysim works, where undriven signals are bound to registers by default.)

BracketMaster, 2020-12-02:
ah. makes sense 👍

whitequark, 2020-12-03:
Fixed upstream.

closed 2020-12-03 by whitequark