amaranth-lang/amaranth #555: Any way to warn about & in if?

RobertBaruch, 2020-12-05:
After having used nMigen for many many months, I've found that one of my most common mistakes is doing this:

```python
        with m.If(self._instr_phase == 0 & ~self._execute_trap):  # Oops, wrong precedence for & vs ==
            m.d.comb += self._load_instr.eq(1)
```

where instead I should do this:

```python
        with m.If((self._instr_phase == 0) & ~self._execute_trap):
            m.d.comb += self._load_instr.eq(1)
```

I undestand why we can't use `and`, because there's no dunder for that. And I also understand that `0 & x` can look just like `0xFFFE & x`, where you meant to do that, so `y == 0 & x` might look the same as `y == 0xFFFE & x`.

I just wish there were some way of telling you that what you did might not be what you wanted. Just planting the thought.

whitequark, 2020-12-05:
#380?

RobertBaruch, 2020-12-05:
Closed as duplicate, will comment further in #380.

closed 2020-12-05 by RobertBaruch