amaranth-lang/amaranth #556: cxxsim: simulator-only signals not included in VCD and GTKWave files

cestrauss, 2020-12-08:
Simulator-only signals (not used in any Module) are useful to me for communicating between parallel processes, say between a main driver and a passive process simulating a RTL module.

They are also useful as a way to display some simulator state on the VCD.

Consider:
```python
from nmigen import Signal, Module
from nmigen.sim import Simulator

m = Module()
s = Signal()
t = Signal()
m.d.sync += s.eq(1)

def process():
    yield t.eq(1)

sim = Simulator(m, engine="cxxsim")
sim.add_clock(1e-6)
sim.add_sync_process(process)
with sim.write_vcd("bug15.vcd", "bug15.gtkw", traces=[t]):
    sim.run()
```

Firstly, Signal ``t`` is absent from the VCD file, unlike pysim.
Secondly, I get the following stack trace:
```
$ python bug15.py
Traceback (most recent call last):
  File "bug15.py", line 18, in <module>
    sim.run()
  File "/usr/lib/python3.7/contextlib.py", line 119, in __exit__
    next(self.gen)
  File "/home/cstrauss/src/nmigen/nmigen/sim/cxxsim.py", line 279, in write_vcd
    gtkw_save.trace(".".join(self._state.names[signal]) + suffix)
  File "/home/cstrauss/src/nmigen/nmigen/hdl/ast.py", line 1561, in __getitem__
    return self._storage[key]
KeyError: <nmigen.hdl.ast.SignalKey (sig t)>
```

whitequark, 2020-12-08:
Ah, this was even a part of #324, but I think I accidentally dropped it at some point.

cestrauss, 2020-12-08:
To clarify, they actually work, they are just not included in the VCD file.

whitequark, 2020-12-08:
Yep, I even know the exact change needed to add them to the VCD files, I just forgot to apply it.

whitequark, 2020-12-12:
Fixed.

commit 08071498 (2020-12-12): sim.cxxsim: dump simulation-only signals to VCD, when possible.

Fixes #556.
commit 547c296a (2020-12-12): sim.cxxsim: dump simulation-only signals to VCD, when possible.

Fixes #556.