amaranth-lang/amaranth #557: cxxsim: Python process can't override combinatorial assignment

cestrauss, 2020-12-08:
Found this by accident, as I don't really have a use case for overriding driven signals.  Nevertheless, pysim seems to allow this. 

Consider:
```python
from nmigen import Signal, Module
from nmigen.sim import Simulator

m = Module()
s = Signal()
m.d.comb += s.eq(1)

def process():
    yield s.eq(0)

sim = Simulator(m, engine="cxxsim")
sim.add_process(process)
sim.run()
```

I get:
```
$ python bug16.py 
Traceback (most recent call last):
  File "bug16.py", line 15, in <module>
    sim.run()
  File "/home/cstrauss/src/nmigen/nmigen/sim/core.py", line 168, in run
    while self.advance():
  File "/home/cstrauss/src/nmigen/nmigen/sim/core.py", line 159, in advance
    return self._engine.advance()
  File "/home/cstrauss/src/nmigen/nmigen/sim/cxxsim.py", line 241, in advance
    self._step()
  File "/home/cstrauss/src/nmigen/nmigen/sim/cxxsim.py", line 231, in _step
    process.run()
  File "/home/cstrauss/src/nmigen/nmigen/sim/_pycoro.py", line 123, in run
    self.coroutine.throw(exn)
  File "/home/cstrauss/src/nmigen/nmigen/sim/core.py", line 84, in wrapper
    yield from process()
  File "bug16.py", line 10, in process
    yield s.eq(0)
  File "/home/cstrauss/src/nmigen/nmigen/sim/_pycoro.py", line 76, in run
    self.exec_locals)
  File "<string>", line 1, in <module>
  File "/home/cstrauss/src/nmigen/nmigen/sim/cxxsim.py", line 38, in next
    value |= part.next
  File "/home/cstrauss/src/nmigen/nmigen/sim/_cxxrtl.py", line 115, in next
    value |= self._next[chunk]
ValueError: NULL pointer access
```

whitequark, 2020-12-08:
I'm fairly certain this doesn't actually work in PySim either (the "override" is silently reverted the next time any inputs to the combinatorial function change), so I would classify the fact that it seems to be alloewd in PySim as a bug.

whitequark, 2020-12-08:
Though, both PySim and CXXSim need to be fixed: the former so that it is actually an error, the latter so that the error is nicer.

commit 122be784 (2024-04-13): sim: raise an error when overriding a combinationally-driven signal.

Fixes #557.
PR #1328 (2024-04-13): sim: raise an error when overriding a combinationally-driven signal.
Fixes #557.
closed 2024-04-13 by whitequark