amaranth-lang/amaranth #560: Rounding errors in vcd simulator time

nturley, 2020-12-12:
As a consequence of #535, there's rounding errors in simulation time. I'm guessing this isn't a huge deal if your timescale is small enough and you are just plotting the result but I was trying to analyze simulation output and I was seeing the clock frequency move around a little bit during the simulation which was confusing.

## To reproduce
I was working through the example [here](https://nmigen.info/nmigen/latest/start.html)

```python
sim = Simulator(dut)
sim.add_clock(1e-6) # 1 MHz
```
result
```
$comment Generated by nMigen $end
$date 2020-12-12 10:12:39.508725 $end
$timescale 100 ps $end
$scope module top $end
$var wire 1 0 ovf $end
$var wire 16 1 count $end
$var wire 1 2 clk $end
$var wire 1 3 rst $end
$var wire 1 4 en $end
$upscope $end
$enddefinitions $end
#0
$dumpvars
00
b0 1
02
03
04
$end
#5000
12
#10000
02
#15000
12
#20000
02
#24999
12
#29999
02
#34999
12
#40000
02
#45000
```

whitequark, 2020-12-12:
Isn't this issue just a duplicate of #535? The rounding errors are the exact problem that issue exists for,

nturley, 2020-12-12:
Fair enough.