amaranth-lang/amaranth #564: cxxsim: directly initializing memory

cestrauss, 2020-12-19:
We currently use code similar to this, in PySim, to re-initialize the memory state in one go, before a sub-test:
```python
from nmigen import Module, Memory
from nmigen.sim import Simulator

m = Module()
mem = Memory(width=64, depth=8)
rdport = mem.read_port()
m.submodules.rdport = rdport

def process():
    yield mem._array[0].eq(0x5432123412345678)
    yield
    # both cxxsim and pysim passes
    assert (yield mem[0]) == 0x5432123412345678
    # only pysim passes
    assert (yield rdport.data) == 0x5432123412345678

for engine in ["pysim", "cxxsim"]:
    sim = Simulator(m, engine=engine)
    sim.add_clock(1e-6)
    sim.add_sync_process(process)
    sim.run()
    print(f"Engine {engine} OK.")
```

I suppose it's not really a CXXSim bug, since we shouldn't really be writing to some private Memory array. More of a feature request, I guess.

whitequark, 2020-12-20:
> I suppose it's not really a CXXSim bug, since we shouldn't really be writing to some private Memory array.

Why are you not using `yield mem[0].eq(0x5432123412345678)` in pysim-only code?

cestrauss, 2020-12-20:
> Why are you not using `yield mem[0].eq(0x5432123412345678)` in pysim-only code?

It was like that in the original test code for some reason. I wrongly assumed it was deliberately working around a nMigen restriction.
Also, I foolishly looked for a ``__setitem__`` in Memory. I now realize I don't really need it, just to call `eq` on it.

Revised test below:
```python
from nmigen import Module, Memory
from nmigen.sim import Simulator

m = Module()
mem = Memory(width=64, depth=8)
rdport = mem.read_port()
m.submodules.rdport = rdport

def process():
    yield mem[0].eq(0x5432123412345678)
    yield
    # both cxxsim and pysim passes
    assert (yield mem[0]) == 0x5432123412345678
    # only pysim passes
    assert (yield rdport.data) == 0x5432123412345678

for engine in ["pysim", "cxxsim"]:
    sim = Simulator(m, engine=engine)
    sim.add_clock(1e-6)
    sim.add_sync_process(process)
    sim.run()
    print(f"Engine {engine} OK.")
```

whitequark, 2020-12-20:
Right, in this case this is a current limitation of cxxsim already mentioned in #324 (but it's ok to track it separately here).