amaranth-lang/amaranth #565: cxxsim: random garbage in memory traces

cestrauss, 2020-12-19:
Consider:
```python
from nmigen import Module, Memory
from nmigen.sim import Simulator

m = Module()
mem = Memory(width=64, depth=8)
rdport = mem.read_port()
m.submodules.rdport = rdport

def process():
    yield

for engine in ["pysim", "cxxsim"]:
    sim = Simulator(m, engine=engine)
    sim.add_clock(1e-6)
    sim.add_sync_process(process)
    with sim.write_vcd(f"bug18.{engine}.vcd"):
        sim.run()
```

The CXXSim VCD file shows some strange behavior:
1) `rdport` sub-module is absent
2) The `top` module has some `mem[<i>][63:0]` signals that appear to contain random garbage.

whitequark, 2020-12-20:
> 1\. `rdport` sub-module is absent

This is expected; `rdport` does not exist in synthesis. If there will be a change related to `rdport`, it will be to remove it from pysim vcd files as well.

> 2\. The `top` module has some `mem[<i>][63:0]` signals that appear to contain random garbage.

Will look into this.

whitequark, 2020-12-20:
Confirmed random garbage; looks ASCII to me, so probably uninitialized memory of some sort?
[image]

whitequark, 2020-12-20:
This is a problem with the way the simulation is reset. It is not entirely trivial to fix.

whitequark, 2020-12-20:
This is fixed by https://github.com/YosysHQ/yosys/pull/2495.

cestrauss, 2020-12-28:
Just to confirm that I've tried the above Yosys branch, and it does seem to fix the random garbage.
Also, I checked that the VCD traces of the memory array do reflect the values written through a write port.
For instance:
```python
from nmigen import Module, Memory
from nmigen.sim import Simulator

m = Module()
mem = Memory(width=64, depth=8)
wrport = mem.write_port()
rdport = mem.read_port()
m.submodules.rdport = rdport
m.submodules.wrport = wrport

def process():
    yield wrport.en.eq(1)
    yield wrport.data.eq(0x5432123412345678)
    yield
    yield

sim = Simulator(m, engine="cxxsim")
sim.add_clock(1e-6)
sim.add_sync_process(process)
with sim.write_vcd("bug27.vcd"):
    sim.run()
```
In the VCD trace, `mem[0][63:0]` does reflect the written value, while the others do remain zeroed.

Not all is well, though. The branch seems to introduce a regression.
Code:
```python
from nmigen import Module, Signal
from nmigen.sim import Simulator

m = Module()
s = Signal()
m.d.sync += s.eq(1)

def process():
    yield

sim = Simulator(m, engine="cxxsim")
sim.add_clock(1e-6)
sim.add_sync_process(process)
sim.run()
```
Result:
```
$ python bug26.py
/home/cstrauss/src/nmigen/nmigen/sim/cxxsim.py:198: YosysWarning: The `-Og` option has been removed. Use `-g3` instead for complete design coverage regardless of optimization level.
  cxx_source, name_map = cxxrtl.convert_fragment(fragment)
sim.cc: In constructor ‘cxxrtl_design::p_top::p_top(cxxrtl::adopt, cxxrtl_design::p_top)’:
sim.cc:55:2: error: expected identifier before ‘{’ token
  {
  ^
Traceback (most recent call last):
  File "/usr/lib/python3.7/distutils/unixccompiler.py", line 118, in _compile
    extra_postargs)
  File "/usr/lib/python3.7/distutils/ccompiler.py", line 909, in spawn
    spawn(cmd, dry_run=self.dry_run)
  File "/usr/lib/python3.7/distutils/spawn.py", line 36, in spawn
    _spawn_posix(cmd, search_path, dry_run=dry_run)
  File "/usr/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
    % (cmd, exit_status))
distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "bug26.py", line 14, in <module>
    sim = Simulator(m, engine="cxxsim")
  File "/home/cstrauss/src/nmigen/nmigen/sim/core.py", line 70, in __init__
    self._engine   = engine(self._fragment)
  File "/home/cstrauss/src/nmigen/nmigen/sim/cxxsim.py", line 210, in __init__
    output_name="sim"
  File "/home/cstrauss/src/nmigen/nmigen/_toolchain/cxx.py", line 51, in build_cxx
    cc_driver.compile(cxx_filenames)
  File "/usr/lib/python3.7/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/lib/python3.7/distutils/unixccompiler.py", line 120, in _compile
    raise CompileError(msg)
distutils.errors.CompileError: command 'x86_64-linux-gnu-gcc' failed with exit status 1

```

whitequark, 2020-12-28:
> The branch seems to introduce a regression.

This regression is something that will be fixed by the changes I will apply before merging. Nevertheless, I wasn't aware of it, so thanks anyway!

whitequark, 2021-12-11:
> This is fixed by [YosysHQ/yosys#2495](https://github.com/YosysHQ/yosys/pull/2495).

That PR has been superseded by https://github.com/YosysHQ/yosys/pull/3105.

whitequark, 2021-12-12:
Fixed upstream, with no Amaranth-specific changes needed.

closed 2021-12-12 by whitequark