amaranth-lang/amaranth #566: Python simulator hangs or throws when trying to drive reset from testbench

hansfbaier, 2020-12-25:
I just learned that nmigen handles reset automatically, so in a simple module, I pulled out my own reset logic and tried 
to use the reset logic from the clock domain. That does not seem to be a problem for verilog generation,
but for the python simulator I tried two approaches, and both failed:
1. I get the resetsignal with ResetSignal() and then try to drive it with in the tests. This should work, because the constructors all default to the "sync" domain. But here I get the error:
```
File "/home/jack/riscv/litex-root/nmigen/nmigen/sim/pysim.py", line 209, in __init__
    self.curr = self.next = signal.reset
AttributeError: 'ResetSignal' object has no attribute 'reset'
```
2. I explicitly create a clock domain and feed it into the simulator, but then the simulator does not seem to terminate
[reset-sim-hangs.tar.gz](https://github.com/nmigen/nmigen/files/5742064/reset-sim-hangs.tar.gz)

whitequark, 2020-12-25:
> 1. I get the resetsignal with ResetSignal() and then try to drive it with in the tests.

This part is a duplicate of https://github.com/nmigen/nmigen/issues/415.

whitequark, 2020-12-25:
> 1. I explicitly create a clock domain and feed it into the simulator, but then the simulator does not seem to terminate

In `sim.add_clock(1e-6, domain=sync)`, the `domain` argument is a clock domain name, not a clock domain object. So it should be `sim.add_clock(1e-6, domain="sync")` instead.

hansfbaier, 2020-12-25:
> > 1. I explicitly create a clock domain and feed it into the simulator, but then the simulator does not seem to terminate
> 
> In `sim.add_clock(1e-6, domain=sync)`, the `domain` argument is a clock domain name, not a clock domain object. So it should be `sim.add_clock(1e-6, domain="sync")` instead.
When I do this, then I get the same error as in the first approach:
`  File "/home/jack/riscv/litex-root/nmigen/nmigen/sim/pysim.py", line 209, in __init__
    self.curr = self.next = signal.reset
AttributeError: 'ResetSignal' object has no attribute 'reset'
`

whitequark, 2020-12-25:
Yes. Please see #415 for a workaround.

hansfbaier, 2020-12-25:
> > 1. I explicitly create a clock domain and feed it into the simulator, but then the simulator does not seem to terminate
> 
> In `sim.add_clock(1e-6, domain=sync)`, the `domain` argument is a clock domain name, not a clock domain object. So it should be `sim.add_clock(1e-6, domain="sync")` instead.

Also, when I wrote the code I looked it up in the source code and both the docstring and the source code seemed to suggest that passing the domain would be an option (Simulator.py):
```
        domain : str or ClockDomain
            Driven clock domain. If specified as a string, the domain with that name is looked up
            in the root fragment of the simulation.
```

whitequark, 2020-12-25:
Hm, you're right. I will look into it.

whitequark, 2021-12-11:
> Also, when I wrote the code I looked it up in the source code and both the docstring and the source code seemed to suggest that passing the domain would be an option (Simulator.py):

Sorry, I was wrong about this--you can indeed pass a domain object, and it works just fine.

> 2\. I explicitly create a clock domain and feed it into the simulator, but then the simulator does not seem to terminate

The issue with this approach is that you end up with two different clock domains called "sync" in the simulation; one automatically added by nMigen when you pass your DUT to the simulator (since the DUT does not itself define a "sync" domain), and another explicitly created by you. To avoid this, add a wrapper:

```
    dut = EdgeToPulse()
    wrapper = Module()
    wrapper.domains.sync = sync = ClockDomain("sync")
    sim = Simulator(wrapper)
```

This causes the same error as your first approach, since it's functionally identical to it.

whitequark, 2021-12-11:
I've added a warning to cover the issue in your second approach. It now shows:
```
edgetopulse-bench-second-approach.py:66: UserWarning: Adding a clock process that drives a clock domain object named 'sync', which is distinct from an identically named domain in the simulated design
  sim.add_clock(1e-6, domain=sync)
```

I consider the issue with the first approach a duplicate of #415.

closed 2021-12-11 by whitequark

hansfbaier, 2021-12-11:
Thank you so much for looking into this! 🙏

commit 7e2b7282 (2021-12-11): sim.core: warn when driving a clock domain not in the simulation.

Closes #566.