amaranth-lang/amaranth #569: ECP5 blockram silently trucates depth?

korken89, 2020-12-29:
Hi, I am playing around with FIFOs to better understand them but I have found a weird thing happening.
When instantiating a FIFO larger than the block RAM size of the FPGA, in my case ECP5 with 18 kBits size, it seems that I am still only getting a single BRAM instantiated.
I would have expected these to be cascaded in this case.

E.g. below I have a 16 bit wide RAM I made at least 16k words deep, this should use about 16 BRAMs.
But I get the following resource usage, 1 BRAM, indicating to me that the depth has been silently truncated.
Is this correct? Or maybe I am using the FIFO wrongly?

```
Info: Device utilisation:
Info: 	       TRELLIS_SLICE:   245/41820     0%
Info: 	          TRELLIS_IO:    14/  365     3%
Info: 	                DCCA:     2/   56     3%
Info: 	              DP16KD:     1/  208     0%   <---------- Only 1 BRAM?!?!
Info: 	          MULT18X18D:     0/  156     0%
Info: 	              ALU54B:     0/   78     0%
Info: 	             EHXPLLL:     1/    4    25%
Info: 	             EXTREFB:     0/    2     0%
Info: 	                DCUA:     0/    2     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:     0/  224     0%
Info: 	            SIOLOGIC:     0/  141     0%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/   14     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%
```

My test code:

```python
from nmigen import Elaboratable, Signal, Module, Repl
from nmigen.build import Platform
from nmigen.lib.fifo import AsyncFIFOBuffered
from ..utils.ecp5pll import ECP5PLL, ECP5PLLConfig

class PllTimer(Elaboratable):
    def elaborate(self, platform: Platform):
        led1 = platform.request("led", 0)
        led4 = platform.request("led", 3)

        timer1 = Signal(25)
        fifo_buf = Signal(16)

        m = Module()

        m.submodules.pll = ECP5PLL([
            ECP5PLLConfig("sync", 25),
            ECP5PLLConfig("fast", 100, error=0),
        ])

        # Some really large depth for the FIFO
        m.submodules.fifo = fifo = AsyncFIFOBuffered(
            width=16, depth=16000, r_domain="fast", w_domain="sync")

        # Write the FIFO using the top 16 bits of the `timer1` data in the `sync` domain
        m.d.sync += timer1.eq(timer1 + 1)
        with m.If(fifo.w_rdy):
            m.d.comb += fifo.w_data.eq(timer1[9:25])
        m.d.comb += fifo.w_en.eq(1)

        # Read the FIFO in the `fast` domain, the LEDs should blink at the same time
        with m.If(fifo.r_rdy):
            m.d.fast += fifo_buf.eq(fifo.r_data)
        m.d.comb += fifo.r_en.eq(1)

        # Connect LEDs to top bits
        m.d.comb += led1.o.eq(timer1[-1])
        m.d.comb += led4.o.eq(fifo_buf[-1])

        return m
```

daveshah1, 2020-12-29:
Looks like you are only using one output bit, so the others may be being swept away by Yosys? Try doing a reduce-xor to use all the bits, for example.

korken89, 2020-12-29:
Ah, that was the issue! I did not expect that Yosys would be that smart :) 
Thanks for clearing that up for me and thanks for the prompt reply!

closed 2020-12-29 by korken89