amaranth-lang/amaranth #572: Add option to suppress code generation comments

hansfbaier, 2021-01-05:
When generating code, almost every line has code generation comments.
While this is useful for debugging, it makes the code very hard to read.
Therefore it would be great to have a code generation option
to suppress those.

hellow554, 2021-02-05:
hey @hansfbaier 
you seem to have been working on this issue (according to the three references commits here).
What is your status on this? Are you able to submit a PR?

hansfbaier, 2021-02-06:
@hellow554 : I have this: https://github.com/hansfbaier/nmigen/commit/304ad1ffa2b64fe60cb709d9dad2cb9edaf9d87c
But I am hesitant to create a pull request, because whitequark seems opposed to having this functionality in nmigen,
see https://github.com/nmigen/nmigen/pull/573

whitequark, 2021-02-06:
I think the functionality is fine, the implementation just needs to be improved. I apologize for my disproportionally intense reaction, @hansfbaier--I'll work something out to have this upstream.

hansfbaier, 2021-02-06:
@whitequark  Thank you!

commit fd7d01ef (2021-12-11): back.rtlil,cli: allow suppressing generation of `src` attributes.

Fixes #572.
closed 2021-12-11 by whitequark