amaranth-lang/amaranth #581: $verilog_initial_trigger leading to `1'hx = 1'hx;` in generated Verilog

adamgreig, 2021-01-24:
I'm using nmigen 6ce2b21e and Yosys 0.9+3521 (git sha1 4f2b78e1, clang 6.0.0-1ubuntu2 -fPIC -Os).

When a signal is assigned to by a single comb statement, I end up with `1'hx = 1'hx;` in the output Verilog, which breaks some other tools (specifically, it works for me with yosys+nextpnr-ecp5, but fails in Diamond). I'm using `LatticeECP5Platform` in the actual code (but the issue is still present in the reproduction below); the generated yosys script does include `delete w:$verilog_initial_trigger`.

Repro:

```python
>>> from nmigen import Module, Signal
>>> from nmigen.back import verilog, rtlil
>>> m = Module()
>>> s = Signal(1)
>>> m.d.comb += s.eq(1)
>>> print(verilog.convert(m))
(* \nmigen.hierarchy  = "top" *)
(* top =  1  *)
(* generator = "nMigen" *)
module top();
  reg \initial  = 0;
  (* src = "<stdin>:1" *)
  reg s;
  always @* begin
    if (\initial ) begin end
    s = 1'h0;
    s = 1'h1;
    1'hx = 1'hx;
  end
endmodule

>>> print(rtlil.convert(m))
attribute \generator "nMigen"
attribute \top 1
attribute \nmigen.hierarchy "top"
module \top
  attribute \src "<stdin>:1"
  wire width 1 \s
  wire width 1 $verilog_initial_trigger
  process $group_0
    assign \s 1'0
    assign \s 1'1
    assign $verilog_initial_trigger $verilog_initial_trigger
    sync init
      update $verilog_initial_trigger 1'0
  end
end
```

adamgreig, 2021-01-24:
Fixed by upgrading yosys, I should know better... sorry for the noise!

```
(* \nmigen.hierarchy  = "top" *)
(* top =  1  *)
(* generator = "nMigen" *)
module top();
  (* src = "<stdin>:1" *)
  wire s;
  assign s = 1'h1;
endmodule
```

closed 2021-01-24 by adamgreig

whitequark, 2021-01-24:
> Fixed by upgrading yosys

There is a Yosys version check guarding that code, so it should have been caught...