amaranth-lang/amaranth #583: Unusual Display Behaviors

BracketMaster, 2021-01-27:
Sometimes display will trigger twice(when it should trigger once) or once(when
it should trigger twice).

Below, I present a case where display should fire twice in state two:

```python3
from nmigen import Module, Display, Signal

m = Module()

sig = Signal()

with m.FSM(name="simple"):
    with m.State("one"):
        m.d.comb += Display("one")
        m.next = "two"
    with m.State("two"):
        m.d.comb += Display("two")
        m.next = "two"

from nmigen.sim import Simulator, Tick

sim = Simulator(m)
sim.add_clock(1e-6)

def process():
    yield Tick()

sim.add_sync_process(process)
sim.run()

```

evaluating this produces:

```bash
$ python3 test.py
one
two
```

where we expect:

```bash
$ python3 test.py
one
two
two
```

A counterexample:

By changing the following

```python3
    with m.State("two"):
        m.d.comb += Display("two")
        m.next = "two"
```

to 

```python3
    with m.State("two"):
        m.d.comb += Display("two")
        m.next = "one"
```

we observe in the expected behavior of

```bash
$ python3 test.py
one
two
one
```

BracketMaster, 2021-01-27:
after further investigation, it seems that using Display within sync is more reliable

whitequark, 2021-01-27:
This is generally expected behavior for comb Display.

BracketMaster, 2021-01-29:
Alright. I think I agree.

closed 2021-01-29 by BracketMaster

whitequark, 2021-01-29:
Once there is a sink for `Display`-produced events I think you will be able to post-process them to make sure only one fires during a given simulation time instant.