amaranth-lang/amaranth #588: Left shift + simulator = infinite (?) loop

hellow554, 2021-02-02:
The following code does not complete (at least not in 1 minute, so I call it infinite ;) )

```python
from nmigen import *
from nmigen.sim import *

class A(Elaboratable):
    def elaborate(self, _):
        m = Module()
        a = Signal(32)
        b = Signal(32)
        z = Signal(32)
        m.d.comb += z.eq(a << b)
        return m

m = Module()
m.submodules.a = A()
sim = Simulator(m)
```

The problem seems to be the `a << b` operation.

rroohhh, 2021-02-02:
@hellow554 It is simply very slow, as `a << b` generates a `4294967327` bit wide intermediate wire. Converting this to verilog / rtlil already catches [this](https://github.com/nmigen/nmigen/blob/746886ca8ac3b9a8941b540a347452805acbbcf2/nmigen/back/rtlil.py#L139-L141). 

Maybe a similar check should be added to the simulator backend.

hellow554, 2021-02-02:
And a right shift is okay, because there are only 32 possible values?
So the solution would be to shorten `b` to e.g. 5 bits?

rroohhh, 2021-02-02:
Yep

whitequark, 2021-02-02:
Yep, the manual has [this note](https://nmigen.info/nmigen/latest/lang.html#lang-hugeshift). We could probably add a similar check to pysim as to the RTLIL backend.

commit ac13a5b3 (2021-12-11): sim._pyrtl: reject very large values.

A check that rejects very large wires already exists in back.rtlil
because they cause performance and correctness issues with Verilog
tooling. Similar performance issues exist with the Python simulator.

This commit also adjusts back.rtlil to use the OverflowError
exception, same as in sim._pyrtl.

Fixes #588.
closed 2021-12-11 by whitequark