amaranth-lang/amaranth #590: How to derive the negative edge from a clock

hellow554, 2021-02-05:
I have a myhdl background and maybe that's why this seems like a very easy task to me.

in myhdl you can define a elaboratele thing like this (http://docs.myhdl.org/en/latest/manual/reference.html#myhdl.always_seq)
```python
@always_seq(clk.negedge, reset=reset)
def logic():
    self.output = self.input
return logic
```

I can't seem to find an easy way to do something similar in nmigen, e.g:

```
m.d.sync.neg += self.output.eq(self.input)
```

What is the idiomatic way to do this?

(Ps, if you open the discussion tab, you can easily move an (/this) issue to discussions, so they don't litter the issue tab ;) )

rroohhh, 2021-02-05:
Currently this is done by creating a new clockdomain, driving it by the inverse of the clocksignal and adding it to your module.

whitequark, 2021-02-05:
Yep, creating a new clock domain is the idiomatic way (arguably, it *is* a new domain--at least if you include the polarity in the control set, which I do). You probably want to drive it with the exact same clock signal though, not the inverse.

rroohhh, 2021-02-05:
Whoops yes, I completely missed the `clk_edge=...` argument of `ClockDomain`, sorry!

hellow554, 2021-02-05:
RobertBaruch has exactly this topic covered in his tutorial:

https://github.com/RobertBaruch/nmigen-tutorial/blob/master/3_modules.md#tip-clock-domains-with-the-same-clock-but-different-edges

I should go a read more of him, thanks everybody.

closed 2021-02-07 by hellow554