amaranth-lang/amaranth #592: Fragment.get should detect cycles in elaboration

Ravenslofty, 2021-02-10:
While watching a stream by @awygle I noticed Fragment.get() has a `while True` loop to chain `elaborate` calls. This could be improved by detecting possible loops in elaboration, in the event that some elaboration-time metaprogramming has a bug that causes a cycle.

For example, the following test case will never complete elaborating:
```python
from nmigen import *
from nmigen.back import verilog

class Example(Elaboratable):
    def elaborate(self, platform):
        return self

e = Example()
print(verilog.convert(e))
```

awygle, 2021-02-10:
Doesn't this run into the halting problem, in the general case? We could do something like a recursion limit.

whitequark, 2021-02-10:
@awygle has a good point.

awygle, 2021-02-10:
We could also detect the specific case @Ravenslofty calls out above - seems like a reasonably common failure mode.

whitequark, 2021-12-11:
@awygle I agree.

commit 599615ee (2021-12-11): hdl.ir: reject elaboratables that elaborate to themselves.

Fixes #592.
closed 2021-12-11 by whitequark