amaranth-lang/amaranth #597: When implementing a Platform, how should I add Python code as part of the build?

RobertBaruch, 2021-02-20:
It'd be nice if I could just add functions to the platform and pass files through it somehow. Is that not the Done Thing? Should I instead just write standalone Python tools and invoke them?

whitequark, 2021-02-20:
I don't really understand the scenario you're describing. Can you add more detail? In general, build outputs are not expected to include any code other than the code contained in the host Python process that produces the build outputs; adding code to build outputs complicates remote builds.

anuejn, 2021-02-20:
I have a somewhat related issue in #515

RobertBaruch, 2021-02-20:
One scenario could be, as part of the build process:

1. Generate the RTLIL from source (done in `TemplatedPlatform.toolchain_prepare`)
2. Run yosys on it to generate an optimized or altered RTLIL
3. Read the RTLIL from Python and modify some cells (say, via pyosys), writing the result to another RTLIL file.
4. Run the resulting RTLIL through other platform-specific standalone tools

I guess it makes sense that this shouldn't be done, considering the requirement for remote build capability. In my scenario, since I've already got pyosys, I could run steps 2 and 3 all in `toolchain_prepare`.

closed 2021-02-20 by RobertBaruch

whitequark, 2021-02-20:
Yes. Altnernatively, you could implement a build tool that is called in the normal way via `$PATH`, in which case it's irrelevant what language the tool is written in.