amaranth-lang/amaranth #600: `AssertionError` when running `Platform.build` with a `Module` using an `Instance`.

cr1901, 2021-03-13:
The following code snippet triggers an assertion failure:

```python
from nmigen import *
from nmigen_boards.tinyfpga_ax2 import *

from nmigen.build import *
from nmigen_boards.resources import *

plat = TinyFPGAAX2Platform()
plat.add_resources([Resource("led", 0, Pins("gpio_0:1"))])

class Test(Elaboratable):
    def __init__(self):
        pass

    def elaborate(self, plat):
        m = Module()
        led = plat.request("led")

        m.submodules += [
            Instance("top", o_o=led)
        ]

        return m

plan = plat.build(Test(), name="actual_top", do_build=False)
```

```sh
$ python mk_build.py
Traceback (most recent call last):
  File "mk_build.py", line 25, in <module>
    plan = plat.build(Test(), name="actual_top", do_build=False)
  File "C:/msys64/home/william/projects/fpga/nmigen/nmigen/nmigen/build/plat.py", line 95, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "C:/msys64/home/william/projects/fpga/nmigen/nmigen/nmigen/build/plat.py", line 166, in prepare
    fragment._propagate_ports(ports=self.iter_ports(), all_undef_as_ports=False)
  File "C:/msys64/home/william/projects/fpga/nmigen/nmigen/nmigen/hdl/ir.py", line 462, in _propagate_ports
    self._prepare_use_def_graph(parent, level, uses, defs, ios, self)
  File "C:/msys64/home/william/projects/fpga/nmigen/nmigen/nmigen/hdl/ir.py", line 436, in _prepare_use_def_graph
    subfrag._prepare_use_def_graph(parent, level, uses, defs, ios, top)
  File "C:/msys64/home/william/projects/fpga/nmigen/nmigen/nmigen/hdl/ir.py", line 428, in _prepare_use_def_graph
    add_defs(value._lhs_signals())
  File "C:/msys64/home/william/projects/fpga/nmigen/nmigen/nmigen/hdl/ir.py", line 396, in add_defs
    assert defs[sig] is self
AssertionError
```

The error is that the `Pins` direction (an `inout`?) of the `led` `Resource` does not match what `Instance` expects (an `output`). I'm not sure why this `AssertionError` is triggered. Is there a good place to `raise` a "driver conflict in `Instance`" error before this assertion triggers?

Fix to stop the `AssertionError`:

```python
plat.add_resources([Resource("led", 0, Pins("gpio_0:1", dir="o"))])
```

awygle, 2021-03-13:
I believe this would be addressed by #596

whitequark, 2021-03-14:
> The error is that the `Pins` direction (an `inout`?)

That's not an `inout`. You're requesting a triple, implicitly. Use `dir="o"` when defining the LED resource, or better, `LEDResources`.

cr1901, 2021-03-14:
#596 will close this.

PR #1104 (2024-02-27): ir: kill Fragment.ports
Fixes #191.
Fixes #320.
Fixes #398.
Fixes #405.
Fixes #600.
closed 2024-02-27 by whitequark