amaranth-lang/amaranth #608: [RFC] get rid of that `+=` and `.eq()`

github-4o, 2021-04-11:
replace `m.d.<domain> += <target>.eq(<val>)` with `m.d.<domain>.<target> = <val>`

implementation as runtime fix:
```python
from nmigen.hdl.dsl import _ModuleBuilderDomain as b

def fix_semantix():
    def mysetattr(self, name, val):
        if name == "_domain":
            self.__dict__[name] = val
        else:
            if name in self._builder.__dict__:
                self.__iadd__(self._builder.__dict__[name].eq(val))
            else:
                raise Exception("couldn't find {} signal on the module".format(name))
    setattr(b, "__setattr__", mysetattr)
```

- why?
- more eye-pleasing cause you are used to it in other languages

pros: none, just a semantics fix
cons: `__setattr__` looks like a dirty hack that can create problems in the future

whitequark, 2021-04-11:
The left-hand side of `.eq()` can be more complex than just a signal name. For example, you can do:

```
m.d.sync += buf.word_select(idx, 4).eq(input)
```

This is not representable with the syntax you're suggesting.

(Aside from that, this proposal violates Python variable scoping in a way far more substantial than what nMigen already does, but that's a secondary issue compared to the one above.)

ildus, 2021-07-12:
@whitequark how about this approach: https://github.com/m-labs/nmigen/issues/345? I created the issue in the wrong repository by mistake.

modwizcode, 2021-12-11:
Personally, we find that this sort of change obscures important details about what's actually occurring when you assign a signal in a domain like this. You're adding a new assignment statement and the code itself should reflect that. Beyond the potential implementation issues, this fact becomes important because later assignments can override earlier assignments that have been added.

whitequark, 2022-04-05:
The proposal as submitted has a fatal issue: it makes a substantial amount of existing code unrepresentable.

The proposal as modified later has a fatal issue: the code style it proposes makes it impossible to reuse the decision trees built through `If`/`Switch` statements to drive both combinatorial and synchronous signals in the same statement body (or at least the same decision tree). This was one of the substantial issues with Migen that Amaranth aimed to address.

closed 2022-04-05 by whitequark