amaranth-lang/amaranth #615: Integrating design with bus.

m-kru, 2021-05-23:
I struggle to find an documentation or tutorial on how to integrate a design with the bus (such as AXI or Wishbone), and whether there is an option to automatically generate status and control registers.
I have found in https://github.com/m-labs/misoc README that there used to be some kind of "automatic CSR maps" but it also misses to point to the source.

Is it possible to automatically generate control/status registers with nmigen?
If yes, is there any tutorial on how to do that?

whitequark, 2021-05-23:
This is handled as a part of https://github.com/nmigen/nmigen-soc/. However, that repository contains only a partial implementation at the moment.

m-kru, 2021-05-23:
@whitequark Looks like there has been no work for few months.

whitequark, 2021-05-23:
There's been a pandemic.

andresdemski, 2021-06-05:
How can we contribute to the project?