amaranth-lang/amaranth #617: Yosys assertion when converting SyncFIFO

anuejn, 2021-05-30:
`SyncFIFO` cant be used with recent yosys anymore.

 Probably related to https://github.com/YosysHQ/yosys/commit/c7076495f197732725456992c9a02aed9966139a ?

repro:
```python
from nmigen import *
from nmigen.back import verilog
from nmigen.lib.fifo import SyncFIFO
verilog.convert(SyncFIFO(width=8, depth=8))
```
results in:
```
nmigen._toolchain.yosys.YosysError: ERROR: Assert `!port.transparent' failed in kernel/mem.cc:293.
```

whitequark, 2021-05-31:
Please report this against Yosys,

anuejn, 2021-06-01:
from the yosys issue:
> By the way, there is no such thing as a "transparent" asynchronous read port, and it never behaved differently from a plain one. It just so happens that a check function meant to trigger on invalid memory configs was added, and caught this.

It seems like we should reflect this in the nMigen Memory / FIFO?

mwkmwkmwk, 2021-06-01:
Fixed on the yosys side, sorry for the mess (this issue will only affect yosys master compiled within the last 9 days). `$memrd` cells without a clock are now once again accepted regardless of the (effectively meaningless) value of the `TRANSPARENT` parameter. `$memrd_v2` will follow more strict rules when it lands, but then it's very likely to have per-port-pair transparency masks instead.

whitequark, 2021-06-01:
We should also set `TRANSPARENT` to whatever Yosys expects for such ports. That code was written a *very* long time ago and IIRC it predates any documentation on `$memrd` cells, which I wrote using the code as reference. So it's not unexpected that the nMigen side of this is kinda wrong.

mwkmwkmwk, 2021-06-01:
Verilog frontend uses 0, hence my expectation in the check.  However, I don't see a point of changing this — I explicitely don't want to break compatibility with old RTLIL inputs and `$memrd` should now be accepted according to the old rules.  And when we migrate to `$memrd_v2` for the the new features, we'll get to deal with transparency masks instead.

whitequark, 2021-06-01:
Alright.

anuejn, 2021-06-08:
closing this as it is fixed upstream :)

closed 2021-06-08 by anuejn