amaranth-lang/amaranth #630: Undriven signal ends up as module port

antonblanchard, 2021-08-25:
If a signal has no driver, it ends up as a port:

```
from nmigen import *
from nmigen.back import verilog

class Test(Elaboratable):
    def __init__(self):
        self.val_out = Signal()

    def elaborate(self, platform):
        m = Module()
        int_sig = Signal(8)
        m.d.comb += self.val_out.eq(int_sig)

        return m

if __name__ == "__main__":
    top = Test()
    with open("test.v", "w") as f:
        f.write(verilog.convert(top))
```

```
module top(int_sig);
  input [7:0] int_sig;
  wire val_out;
  assign val_out = int_sig[0];
endmodule
```

This is pretty confusing. Reporting an error here would be great.

antonblanchard, 2021-08-28:
Thanks @BracketMaster, that does fix the issue at the top level.

I think there are a couple of usability issues here:
- If inferring ports in Verilog generation doesn't work then `ports=` shouldn't be an optional argument.
- The actual issue was an undriven signal many layers deep in the hierarchy. It was confusing to see this signal in a bunch of internal module ports when looking at waveforms. I presume every undriven signal makes it's way up to the top level.

whitequark, 2021-12-13:
Let's deprecate optional `ports=` in 0.3 and remove it in 0.4.

PR #658 (2021-12-13): back.{rtlil,verilog}: deprecate implicit ports.
Fixes #630.
commit d83c4a1b (2021-12-13): back.{rtlil,verilog}: deprecate implicit ports.

Fixes #630.
closed 2021-12-13 by whitequark