amaranth-lang/amaranth #635: empty signal slice generates verilog which Quartus cannot parse

hansfbaier, 2021-09-28:
```python
from nmigen import *

class QuartusHiccup(Elaboratable):
    def __init__(self):
        self.out = Signal()

    def elaborate(self, platform):
        m = Module()
        data = Signal(1)
        m.d.sync += self.out.eq(data[1:])
        return m
```
This will generate verilog containing:
`  assign \$1  = + ""; `
which is valid verilog, which the quartus parser cannot read.

hansfbaier, 2021-09-28:
Also the CSR register multiplexer from nmigen_soc seems to generate similar results,
but I have not been able to reduce that yet. I am not sure whether an empty signal slice has something to do with that.

whitequark, 2021-12-11:
This is actually an upstream issue; Yosys should not be generating empty strings in this context. I've fixed one instance of this bug in https://github.com/YosysHQ/yosys/pull/1203 (in 2019!), but it turns out there was another one, which I've fixed just now in https://github.com/YosysHQ/yosys/pull/3103.

Your testcase synthesizes fine on Quartus with Yosys 0.10, which is required as of the current `main` branch, so I'm going to close this issue. I'm not sure whether there are other inputs that will generate the RTLIL that causes the pathological Verilog output; please comment on this issue if you do have such a testcase.

closed 2021-12-11 by whitequark

hansfbaier, 2021-12-11:
Wow, that's awesome! Many thanks!