amaranth-lang/amaranth #637: `sim` submodule is missing after `pip install --upgrade nmigen[builtin-yosys]`

chipmuenk, 2021-09-29:
I'd like to integrate nmigen into a software package of mine which is installed via pip, however, when I do

    pip install --upgrade nmigen[builtin-yosys]

or simply

    pip install nmigen

the `sim` submodule is missing after installation. I tried uninstalling and reinstalling and clearing the pip cache several times but the only thing that worked so far is installing the very latest version

    pip install git+https://github.com/nmigen/nmigen.git#egg=nmigen[builtin-yosys]

instead (which I don't want to do inside a software package). I'm using Windows with Python 3.8.2 running in an venv.

Previously, I had migen installed in the same venv, maybe there is some dependency?

whitequark, 2021-09-29:
At the moment I would recommend using nmigen from git. We'll do a release soon (before the end of 2021), so this won't be necessary for long.

mattvenn, 2021-11-22:
would suggest updating the docs to encourage install from git.

mattvenn, 2021-11-22:
can confirm that installing from git works

whitequark, 2021-11-22:
> would suggest updating the docs to encourage install from git.

The docs already explain that installing from git will result in more recent code: https://nmigen.info/nmigen/latest/install.html#installing-nmigen. Going beyond that is not something I think makes a lot of sense.

chipmuenk, 2021-11-22:
Unfortunately, I haven't managed to add a git-based install to the `requirements.txt` list of my project. As a consequence, I'm using my own fixpoint library for simulation in my project now and will switch to a templating approach for code generation.

closed 2021-11-22 by chipmuenk