amaranth-lang/amaranth #639: Consider issuing a warning when a bare integer is used in Cat() or Repl()

whitequark, 2021-10-02:
In general, using bare integers in bit vector context is error-prone because the width of the integer, when cast to a `Value`, will change with its value. Values other than `0` and `1` are very likely to indicate a potential issue. Whether we want to allow `0` and `1` is an open question, but I suspect that people are relying on that.

adamgreig, 2021-10-02:
I've definitely used `0` and `1` inside `Cat` a bunch for adding an extra bit, in situations [like the UART example](https://github.com/nmigen/nmigen/blob/master/examples/basic/uart.py#L43) where it's convenient for shift registers, but it's not that bad to use `Const` either. I think the meaning is pretty clear with `0` and `1` though.

I checked and about 10% of my `Cat` calls use a bare `0` or `1`.

whitequark, 2021-10-02:
Okay, definitely no warning on bare `0` or `1`.

commit 44b8bd29 (2021-12-11): hdl.ast: warn on bare integer value used in Cat()/Repl().

Fixes #639.
closed 2021-12-11 by whitequark