amaranth-lang/amaranth #641: Attributes are not properly applied to IBUFDS (and possibly other primitives) in Xilinx platform

H-S-S-11, 2021-10-09:
https://pastebin.com/qvymcNWi

In the example an `IBUFDS` resource is created and requested, with particular Attributes:
```python
Resource("IBUFDS", 0,
                DiffPairs("48", "46", conn=("gpio", 1), dir="i"),
                Attrs(IOSTANDARD="LVDS_25", IBUF_LOW_PWR="FALSE")
            ),
```
When run the attributes appear in the generated ucf, and an IBUFDS primitive is instantiated in the generated verilog. However when ISE actually runs synthesis, the instance properties do not show these attributes applied:
[image]
In order for ISE to apply the attributes, they need to be parameters to the IBUFDS which is `Instance`d in xilinx.py.

If the generated verilog is edited where the IBUFDS is Instanced, then the attributes are properly applied:
```verilog
IBUFDS #(
  .IOSTANDARD("LVDS_25"),
  .IBUF_LOW_PWR("FALSE")
) IBUFDS_0_0 (
  .I(IBUFDS_0__p),
  .IB(IBUFDS_0__n),
  .O(IBUFDS_0__i)
);```

![image](https://user-images.githubusercontent.com/52609833/136662602-28b679ed-002e-44fe-8cbd-e62af16686fe.png)

whitequark, 2021-12-13:
Are you sure this is issue is actually present?

The `IBUF_LOW_PWR` attribute, when applied as a parameter to the `IBUFDS` cell in your design, does not change the bitstream whether its value is `"TRUE"` or `"FALSE"`, so looking at it is not informative. However, other attributes, like `DIFF_TERM` or `IOSTANDARD`, are indeed applied when specified in the UCF file, which you can see by looking at the `top_par_pad.txt` report.

whitequark, 2024-02-11:
No response from reporter for over two years.

closed 2024-02-11 by whitequark