amaranth-lang/amaranth #648: Driving different bits of multibit signal from different clock domains results in driver-driver conflict

bl0x, 2021-12-09:
Minimal code:

```
from nmigen import *
from nmigen.cli import main

class MultibitDriverConflict(Elaboratable):
    def elaborate(self, platform):
        a = Signal(2)
        m = Module()
        m.domains += [ ClockDomain("c1"), ClockDomain("c2") ]
        m.d.c1 += a[0].eq(1)
        m.d.c2 += a[1].eq(1)
        return m

if __name__ == "__main__":
    test = MultibitDriverConflict()
    main(test)
```

Output:

```
$ python driver-driver-multibit-signal.py generate -t v
nmigen.hdl.dsl.SyntaxError: Driver-driver conflict: trying to drive (sig a) from d.c2, but it is already driven from d.c1
```

One would expect that the bits of Signal `a` can be driven independently.

bl0x, 2021-12-09:
It is fine, if I declare a like so:

```
a = [Signal() for _ in range(2)]
```

whitequark, 2021-12-10:
Yes, this is expected; see [this section](https://nmigen.info/nmigen/latest/lang.html#assignment-domains) of the manual.

bl0x, 2021-12-10:
Ah, yes. I get it: one signal, one domain. I still tend to think about single bits as independent individuals.

Thanks for pointing to the manual. This can be closed.

closed 2021-12-10 by whitequark