amaranth-lang/amaranth #655: Determine frequency of Cyclone V clock

modwizcode, 2021-12-13:
This needs to actually be measured on real hardware, value is not well documented.

whitequark, 2021-12-13:
The [Cyclone-V handbook](https://www.mouser.com/pdfDocs/cv_5v2.pdf) says:

> If you use the internal oscillator, you can choose a 12.5, 25, 50, or 100 MHz clock under the Device and Pin Options
dialog box, in the Configuration page of the Intel Quartus Prime software.

It's not entirely clear where this adds a divider in the clocking path, so this needs to be checked.

Ravenslofty, 2021-12-13:
The setting in question controls `set_global_assignment -name ACTIVE_SERIAL_CLOCK FREQ_{100,50,25,12_5}MHZ`, which I think *actually* controls the oscillator frequency when loading bitstreams from flash.

To actually check this I think one would need to manually instantiate a `cyclonev_oscillator` block.

Ravenslofty, 2022-11-13:
It seems that [AN 496](https://www.intel.com/content/www/us/en/docs/programmable/683653/current/using-the-internal-oscillator-ip-core.html) documents the Cyclone V internal oscillator at 100 MHz maximum. This is... still incredibly vague but it's better than nothing.