amaranth-lang/amaranth #669: cxxsim: error: ‘p_sub_2e_s’ was not declared in this scope

cestrauss, 2021-12-22:
Consider the following test case:

```python
from amaranth import Signal, Module
from amaranth.sim import Simulator

m = Module()
sub = Module()
m.submodules.sub = sub
s = Signal()
t = Signal()
sub.d.comb += t.eq(s)

sim = Simulator(m, engine="cxxsim")
sim.run()
```

When run, it fails for me with:
```
sim.cc: In member function ‘virtual void cxxrtl_design::p_top::reset()’:
sim.cc:38:2: error: ‘p_sub_2e_s’ was not declared in this scope
  p_sub_2e_s = value<1>{0u};
  ^~~~~~~~~~
```

Full log attached.
[bug28.log](https://github.com/amaranth-lang/amaranth/files/7758976/bug28.log)

Versions:
Amaranth: cxxsim branch (1c0cf92888a16e774768b18fb633ba59a0145704)
Yosys 0.12+40 (git sha1 229980d66, clang 7.0.1-8+deb10u2 -fPIC -Os)
OS: Debian 10

whitequark, 2021-12-25:
Should be fixed by https://github.com/YosysHQ/yosys/pull/3127.

cestrauss, 2021-12-25:
Confirmed fixed, thanks.

closed 2021-12-25 by cestrauss