amaranth-lang/amaranth #670: Imports missing from "nmigen.cli" compatibility wrapper: rtlil, verilog

cestrauss, 2021-12-22:
```
$ python 
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from amaranth.cli import rtlil, verilog
>>> from nmigen.cli import rtlil
__main__:1: DeprecationWarning: instead of nmigen, use amaranth
__main__:1: DeprecationWarning: instead of nmigen.cli, use amaranth.cli
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'rtlil' from 'nmigen.cli' (/home/cstrauss/src/nmigen/nmigen/cli.py)
>>> from nmigen.cli import verilog
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'verilog' from 'nmigen.cli' (/home/cstrauss/src/nmigen/nmigen/cli.py)
```
Amaranth version: commit b1f5664b05725676cafa6c9313096c6fba0a47be (main)

whitequark, 2021-12-22:
So these are just `nmigen.back.verilog`; they were never intentionally exported from `nmgien.cli`, and so aren't present in the wrapper.

There are probably hundreds of other names like that. I consider this a substantial weakness of the Python import system; the module interface and the module implementation are conflated, which makes it quite hard to maintain backwards compatibility as people end up depending on unintentionally exported names,

cestrauss, 2021-12-22:
Got it, thanks.

closed 2021-12-22 by cestrauss