amaranth-lang/amaranth #671: cxxsim: ValueError: 4 is not a valid cxxrtl_type

cestrauss, 2021-12-25:
This code fails for me:
```python
from amaranth import Signal, Module
from amaranth.sim import Simulator

m = Module()
s = Signal()
t = Signal()

m.d.sync += s.eq(0)
with m.If(s):
    m.d.comb += t.eq(1)

def process():
    yield t

sim = Simulator(m, engine="cxxsim")
sim.add_clock(1e-6)
sim.add_sync_process(process)
sim.run()
```
Final error message is: `ValueError: 4 is not a valid cxxrtl_type`
[Full log attached](https://github.com/amaranth-lang/amaranth/files/7776297/bug30.log)

Tested with:
Amaranth cxxsim branch (1c0cf92888a16e774768b18fb633ba59a0145704)
Yosys 0.12+42 (git sha1 7407a7f3e, clang 7.0.1-8+deb10u2 -fPIC -Os) (main branch)
Yosys 0.12+30 (git sha1 fc049e84a, clang 7.0.1-8+deb10u2 -fPIC -Os) (cxxrtl-no-reset-elided branch)

ekliptik, 2022-01-05:
Same issue with my design: https://gitlab.com/tywonemi-school-stuff/risc-je-zisk/-/tree/cxxsim

[Exact error](https://github.com/amaranth-lang/amaranth/files/7814529/log.txt)

whitequark, 2022-01-05:
I'll have this fixed shortly.