amaranth-lang/amaranth #672: Amaranth FSM does not allow asynchronous reset

CoreRasurae, 2021-12-27:
I tried to generate the Verilog code for the following Amaranth FSM with Async reset (BasicFSM.py), with: 
`python BasicFSM.py -t v > toplevel.v`
and I get:
```
   ...
    raise YosysError(stderr.strip())
nmigen._toolchain.yosys.YosysError: ERROR: Multiple edge sensitive events found for this signal!
```
Is this the expected?

The code is:
```
class BasicFSMDevice(Elaboratable):
    def __init__(self):
        self.clk = Signal()
        self.csn = Signal()
        
    def elaborate(self, platform : Platform):
        m = Module()
        
        m.domains += ClockDomain("asyncFSM", async_reset=True, local=True)
        
        m.d.comb += ClockSignal('asyncFSM').eq(self.clk)
        m.d.comb += ResetSignal('asyncFSM').eq(self.csn)
        
        with m.FSM(domain='asyncFSM', reset='IDLE', name='serialFSM'):
            with m.State('IDLE'):
                m.next = 'BUSY'
            with m.State('BUSY'):
                m.next = 'BUSY'
        
        return m
        
    def ports(self) -> List[Signal]:
        return [
            self.clk,
            self.csn
        ]

if __name__ == "__main__":
    simulate = False

    parser = main_parser()
    args = parser.parse_args()

    m = Module()
    m.submodules.basicFSM = basicFSM = BasicFSMDevice()
    
    main_runner(parser, args, m, ports=basicFSM.ports())
```

whitequark, 2021-12-27:
This should be fixed by https://github.com/amaranth-lang/amaranth/pull/667.

CoreRasurae, 2021-12-31:
Yes, the patch for #667 does fix the problem. Thank you!

commit 5a4d45b5 (2022-01-01): back.rtlil: avoid sync process emission in RTLIL.

Avoiding emission of sync processes in RTLIL allows us to avoid a dependency on
matching the behavior expected by Yosys, which generally expects sync processes
in RTLIL to match those emitted by the output from its own Verilog parser.
This also simplifies the logic used in emitting RTLIL overall.

Combinatorial processes are still emitted however. Without these the RTLIL does
not have a high-level understanding of Switch statements, which significantly
diminishes the quality of emitted Verilog, as these are converted to `$mux`
cells in Yosys,  [... cut]
closed 2022-01-01 by whitequark