amaranth-lang/amaranth #675: Add option to export only standards compliant VCD

jeremyherbert, 2022-01-03:
It has been pointed out to me that exporting strings in VCD files is not actually part of the VCD file format (as defined by Verilog 2005). For example, see https://github.com/ghdl/ghdl/issues/647

Would it be acceptable to add an (off by default) flag to the VCD writer to only export a standards compliant VCD file? So the VCD data can be used more broadly in applications that do not support string types.

whitequark, 2022-01-03:
Unfortunately, there is no such thing as "standards compliant VCD"; only "VCD that people want to feed to tools they use". For example, CXXRTL used to export completely "standards compliant VCD", which crashed GTKWave, and this had to be fixed (in an incredibly awkward way) in this PR: https://github.com/YosysHQ/yosys/pull/2881.

The closest thing to what you're asking that actually makes sense to implement is "VCD dialects" (so a GTKWave dialect, a Sigrok dialect, and so on), but I'm unconvinced that this is worth the effort.

jeremyherbert, 2022-01-03:
Thanks for the response. I suppose I’m stuck between a rock and a hard place on this one then. I agree on the dialects; I don’t think it is worth it to implement this.

I will just write a post-processing script to strip out the strings from the VCD. Please feel free to close this.

whitequark, 2022-01-03:
Thank you for understanding.

I feel like someone adding the code to Sigrok to skip strings would be ideal, but unfortunately I don't currently have time to do this.

closed 2022-01-03 by whitequark

jeremyherbert, 2022-01-04:
@whitequark seeing as you are almost certainly more experienced on this topic, could you comment on this commit? https://repo.or.cz/libsigrok/gsi.git/commitdiff/9a2955ec85e6f202fe71dbde19191f7eb53fcacc

Bug tracker is here: https://sigrok.org/bugzilla/show_bug.cgi?id=1757

It appears that Gerhard in sigrok is open to adding/adjusting something, but would like to have some sort of constrained scope of what is valid.

whitequark, 2022-01-06:
> https://repo.or.cz/libsigrok/gsi.git/commitdiff/9a2955ec85e6f202fe71dbde19191f7eb53fcacc

This commit looks fine.

Regarding this:

> System Verilog is said to support strings but would not discuss how to represent them in VCD files. The IEEE standard is said to not be open. Hmm ...

Indeed, there's nothing in IEEE 1800-2017 about dumping strings to VCD files.