amaranth-lang/amaranth #688: Dependency issue: "ImportError: cannot import name 'soft_unicode' from 'markupsafe'"

rossmotley, 2022-03-30:
I installed amaranth for the first time, and encountered this error:

> ImportError: cannot import name 'soft_unicode' from 'markupsafe'

I found the [same error](https://github.com/aws/aws-sam-cli/issues/3661) which says that it's due to a change in [version 2.1.0 of markupsafe](https://markupsafe.palletsprojects.com/en/2.1.x/changes/#version-2-1-0).

I fixed it locally by changing to an older version with `pip install markupsafe==2.0.1`.

Stack trace:

```
...snip...
  File "/Users/rossmotley/Library/Python/3.8/lib/python/site-packages/amaranth/build/__init__.py", line 3, in <module>
    from .plat import *
  File "/Users/rossmotley/Library/Python/3.8/lib/python/site-packages/amaranth/build/plat.py", line 6, in <module>
    import jinja2
  File "/Users/rossmotley/Library/Python/3.8/lib/python/site-packages/jinja2/__init__.py", line 12, in <module>
    from .environment import Environment
  File "/Users/rossmotley/Library/Python/3.8/lib/python/site-packages/jinja2/environment.py", line 25, in <module>
    from .defaults import BLOCK_END_STRING
  File "/Users/rossmotley/Library/Python/3.8/lib/python/site-packages/jinja2/defaults.py", line 3, in <module>
    from .filters import FILTERS as DEFAULT_FILTERS  # noqa: F401
  File "/Users/rossmotley/Library/Python/3.8/lib/python/site-packages/jinja2/filters.py", line 13, in <module>
    from markupsafe import soft_unicode
ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/Users/rossmotley/Library/Python/3.8/lib/python/site-packages/markupsafe/__init__.py)
```

modwizcode, 2022-03-30:
Yes, I’ve encountered this recently too. Are you installing Amaranth with pip or with poetry?

dragonmux, 2022-03-30:
@lethalbit hit this issue yesterday and it turned out to be caused by using an out of date Jinja2 - if you're using Jinja2 versions less than 3 then you'll hit this.

modwizcode, 2022-03-30:
The issue is that if you upgrade to Jinja2 3.1.0 or higher this will break Amaranth itself as the API changes made in that version rename features used in amaranth.build

rossmotley, 2022-03-30:
@modwizcode I installed via pip, and @dragonmux I got `Jinja2` at version `2.11.3`.

dragonmux, 2022-03-30:
You want Jinja2 version 3.0.3 - this will work correctly and without error

modwizcode, 2022-03-30:
The correct fix is likely to fix the use of deprecated structures in Amaranth and upgrade the constraint on Jinja2 if possible. Otherwise Amaranth will have to fix a dependency on a specific version of markupsafe to avoid letting the constraint be filled with a newer incompatible version. 

You hit this issue immediately with poetry because (mostly a guess) it’s constraint solver seems to try to solve for latest versions and pip seems to go for the earliest allowed versions.

PR #689 (2022-03-30): Fix Jinja2 dependency
This PR fixes #688 by upgrading the Jinja2 dependency to minimum of major version 3, which does not have the issue with it's markupsafe dependency being  incorrect. In order to do this, update the uses of `@jinja2.contextfunction` to use the non-deprecated `@jinja2.pass_context` name instead, as this name will be removed in Jinja2 versions `>=3.1.0`.
closed 2022-03-30 by whitequark