amaranth-lang/amaranth #692: Remove the Migen compatibility layer

whitequark, 2022-04-05:
It has (technically) been deprecated since day 1, and hasn't seen much uptake. This interface was considered important for migration of existing codebases, but the (intentional and unintentional) mismatches with Migen behavior meant that it only had a limited usefulness; most code that needed Migen compatibility combines the LiteX and the Amaranth parts of the design through Verilog.

Removing the compatibility layer allows a number of cleanups: the AST visitors can dispatch faster using `type() is` rather than `isinstance()`, and several AST types can be marked `@final`.

It also deemphasizes the relation to Migen. Many of the remaining design choices in Amaranth that directly derive from Migen (records, fragments, AST transformations, board DSL, ...) are load bearing liabilities and there's no reason to have a gratuitous one on top of that.

whitequark, 2023-02-16:
I think we won't be able to do this in 0.4 because Glasgow still uses `amaranth.compat` extensively. Should be fine in 0.5.

whitequark, 2023-07-07:
I've talked with @enjoy-digital and as a result I decided to delay removal of the compatibility layer until we decide on a path forward for LiteX. See https://github.com/enjoy-digital/litex/issues/1727.

whitequark, 2023-07-27:
Discussed this on a call with Rob Taylor and @enjoy-digital today. The plan is to spin off `amaranth.compat` as a separate repo under the LiteX organization and then hand it off to Florent for testing and maintenance. Once that is done the compatibility layer can be removed from Amaranth as it was originally planned.

whitequark, 2023-11-25:
After investigation, it appears that the current implementation of the compatibility layer is essentially impossible to separate from the core of Amaranth, and it will need to be rewritten essentially from scratch using the extensibility facilities Amaranth provides (`ValueCastable`, `ShapeCastable`, etc).

@enjoy-digital We have some important work blocked on the removal of the compat layer (primarily, #949, required for native Amaranth streams), so I will proceed with removal of `amaranth.compat` in Amaranth 0.5. I apologize for any inconvenience this causes.

enjoy-digital, 2023-11-25:
@whitequark: Thanks your care on this and letting me know. I don't want to block your work with this, so yes, please make sure do what is the best for Amaranth.

whitequark, 2023-11-25:
@enjoy-digital As previously discussed, I am still quite happy to supervise the work of an external contributor who will produce a replacement compatibility layer specific for and developed according to LiteX's needs, and to make sure the core Amaranth language includes the necessary hooks to enable the existence of such a layer.

PR #840 (2023-12-13): (Merge after 0.4 is released) Remove the Migen compatibility layer
Fixes #692.

This will stay a draft until a resolution on #692 is reached.
closed 2023-12-13 by whitequark