amaranth-lang/amaranth #703: [RFC] Simulation : abort simulation when step() does not converge

sporniket, 2022-04-21:
# Context

Done with writing my code and running a simulation, the simulator is stuck in an infinite loop. After the last yield of the added sync process, `PySimEngine.step()` is stuck in `while not converged`.

# Summary

* Specify a threshold to the number of iteration to converge in a step, with a sensible default value (1000 ? 10000 ?)
* When the number of iteration to converge goes over the threshold
  * break out of the loop
  * flush the vcd_writers, so that one can do analysis
  * raise an Error, with the timestamp of the simulator

whitequark, 2022-04-21:
In general, if the simulator doesn't converge, it means an issue with the input netlist. I'd say that the appropriate way to handle this is to reject such netlists before they can be simulated, since by the time the simulator is stuck, the original issue (a logic loop of some kind) becomes very hard to find.

sporniket, 2022-04-21:
I understand. Then, as a failsafe, I would not mind to still have a capped number of iterations for the cases that the issue in the netlist is not recognized.

whitequark, 2023-02-06:
I think this doesn't need an RFC, it's a simple internal change to the engine. We don't even have to make the configuration mechanism public.

sporniket, 2023-02-08:
Ok, I will try to do something by next week.