amaranth-lang/amaranth #713: [amaranth-boards] Jinja2 dependency seems to require pinning of MarkupSafe dependency

sporniket, 2022-09-19:
After having looked [the raw logs of CI](https://pipelines.actions.githubusercontent.com/serviceHosts/514a3656-e17d-4829-9e26-0d12794b3619/_apis/pipelines/1/runs/259/signedlogcontent/2?urlExpires=2022-09-19T03%3A32%3A26.4017497Z&urlSigningMethod=HMACV1&urlSignature=jSEyzozlaeoTeSSI26fy2JFvw89st9D4ifa1iiTihBM%3D) for [my PR to amaranth-boards](https://github.com/amaranth-lang/amaranth-boards/pull/204), it seems that the jinja2 version used in amaranth relies on an old Markupsafe version (2.0.1), the newer ones (2.1+) have renamed some of their component (here `soft_unicode`).

Looking at github issues comments on [Jinja2](https://github.com/pallets/jinja/issues/1585#issuecomment-1043892628) and [Markupsafe](https://github.com/pallets/markupsafe/issues/282#issuecomment-1043899655), one is expected to either pin the markupsafe dependency, or update to latest jinja2.

```
2022-09-18T10:15:40.7447907Z ======================================================================
2022-09-18T10:15:40.7448394Z ERROR: pmod (unittest.loader._FailedTest)
2022-09-18T10:15:40.7449178Z ----------------------------------------------------------------------
2022-09-18T10:15:40.7449581Z ImportError: Failed to import test module: pmod
2022-09-18T10:15:40.7449907Z Traceback (most recent call last):
2022-09-18T10:15:40.7464918Z   File "/opt/hostedtoolcache/Python/3.10.6/x64/lib/python3.10/unittest/loader.py", line 154, in loadTestsFromName
2022-09-18T10:15:40.7465404Z     module = __import__(module_name)
2022-09-18T10:15:40.7466121Z   File "/home/runner/work/amaranth-boards/amaranth-boards/amaranth_boards/extensions/pmod.py", line 3, in <module>
2022-09-18T10:15:40.7466535Z     from amaranth.build import *
2022-09-18T10:15:40.7467182Z   File "/home/runner/work/amaranth-boards/amaranth-boards/.eggs/amaranth-0.3-py3.10.egg/amaranth/build/__init__.py", line 3, in <module>
2022-09-18T10:15:40.7467587Z     from .plat import *
2022-09-18T10:15:40.7468207Z   File "/home/runner/work/amaranth-boards/amaranth-boards/.eggs/amaranth-0.3-py3.10.egg/amaranth/build/plat.py", line 6, in <module>
2022-09-18T10:15:40.7468792Z     import jinja2
2022-09-18T10:15:40.7469399Z   File "/home/runner/work/amaranth-boards/amaranth-boards/.eggs/Jinja2-2.11.3-py3.10.egg/jinja2/__init__.py", line 12, in <module>
2022-09-18T10:15:40.7470086Z     from .environment import Environment
2022-09-18T10:15:40.7470777Z   File "/home/runner/work/amaranth-boards/amaranth-boards/.eggs/Jinja2-2.11.3-py3.10.egg/jinja2/environment.py", line 25, in <module>
2022-09-18T10:15:40.7471247Z     from .defaults import BLOCK_END_STRING
2022-09-18T10:15:40.7471914Z   File "/home/runner/work/amaranth-boards/amaranth-boards/.eggs/Jinja2-2.11.3-py3.10.egg/jinja2/defaults.py", line 3, in <module>
2022-09-18T10:15:40.7472412Z     from .filters import FILTERS as DEFAULT_FILTERS  # noqa: F401
2022-09-18T10:15:40.7473101Z   File "/home/runner/work/amaranth-boards/amaranth-boards/.eggs/Jinja2-2.11.3-py3.10.egg/jinja2/filters.py", line 13, in <module>
2022-09-18T10:15:40.7473762Z     from markupsafe import soft_unicode
2022-09-18T10:15:40.7474577Z ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/home/runner/work/amaranth-boards/amaranth-boards/.eggs/MarkupSafe-2.1.1-py3.10-linux-x86_64.egg/markupsafe/__init__.py)
```

sporniket, 2022-09-20:
> [@whitequark said](https://github.com/amaranth-lang/amaranth/pull/714#issuecomment-1251818357) :
> I don't think this is an appropriate fix? The `amaranth` package shouldn't be introducing dependencies to fix issues with its dependent `amaranth-boards` package. I'll take a closer look at this later if you can't improve on this PR.

Then I'll try to look into the amaranth_board too.

whitequark, 2022-09-24:
So this isn't actually an issue with the `main` branch as here the requirement in `setup.py` is `Jinja2~=3.0`.

closed 2022-09-24 by whitequark