amaranth-lang/amaranth #718: [amaranth-boards] make a release with fixed jinja2 dependency

sporniket, 2022-09-25:
Follow up of #713 

Trying various things on amaranth-board project, I stumbled upon [this comment](https://github.com/amaranth-lang/amaranth-boards/issues/200#issuecomment-1100476052) mentioning issue #688, that has been solved with PR #689

However, amaranth-board is using amaranth release 0.3 (it requires a version <0.5) , that is anterior to that fix. 

I believe that publishing on pypi a hotfix of 0.3 that only include #689 would solve this particular problem of amaranth-board (and eventually unmask other problems that are hidden by this dependency problem)

sporniket, 2022-09-25:
Another option would be to regularly publish (monthly or every other month) amaranth to pypi as it is, considering a twisted semver `0.Major.Minor.Fix`, so that we stay at _v0.3..._ until _v0.4_ is ready

sporniket, 2022-10-02:
Me again, I also succeeded to fix the CI on amaranth-board by referencing a recent commit of this github repository in setup.py instead of the package on amaranth, see PR https://github.com/amaranth-lang/amaranth-boards/pull/205 . If this is more manageable/acceptable because it requires no action from the side of amaranth project.

sporniket, 2022-10-06:
See https://github.com/amaranth-lang/amaranth-boards/issues/200#issuecomment-1269662610

daquintero, 2023-08-17:
Hi! Is there any chance this could be released? It's beginning to conflict with common tools like JupyerLab

whitequark, 2023-08-17:
This will be released as a part of Amaranth 0.4.

daquintero, 2023-08-21:
I tried to follow this same principle when adding `amaranth` as a dependency for my project, and worth mentioning it doesn't seem this syntax in the requirements will upload to PyPi just in case this comes up for the future for you.

```
Checking dist/piel-0.0.49-py2.py3-none-any.whl: PASSED
Checking dist/piel-0.0.49.tar.gz: PASSED
Uploading distributions to https://upload.pypi.org/legacy/
Uploading piel-0.0.49-py2.py3-none-any.whl
25l
  0% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 0.0/89.7 kB • --:-- • ?
  0% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 0.0/89.7 kB • --:-- • ?
100% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.7/89.7 kB • 00:00 • 2.0 MB/s
100% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.7/89.7 kB • 00:00 • 2.0 MB/s
100% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.7/89.7 kB • 00:00 • 2.0 MB/s
100% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.7/89.7 kB • 00:00 • 2.0 MB/s
[25](https://github.com/daquintero/piel/actions/runs/5932203509/job/16085574828#step:7:26)hWARNING  Error during upload. Retry with the --verbose option for more details. 
ERROR    HTTPError: 400 Bad Request from https://upload.pypi.org/legacy/        
         Invalid value for requires_dist. Error: Can't have direct dependency:  
         'amaranth @ git+https://github.com/amaranth-lang/amaranth.git'
```

Trying to think of an alternative solution atm.

The pypi issue is detailed here https://github.com/pypa/twine/issues/726

Part of me thinks a automatic bumpversion-github action versioned system based on the main branch being uploaded to pypi like `gdsfactory` does would solve the problem https://gdsfactory.github.io/gdsfactory/ as it means most users can use the major stable versions, but any latest build versions can also be version controlled. But looking for a solution in the interim

whitequark, 2023-08-22:
There is no solution that you can use at the moment. You can track the progress to the Amaranth 0.4 release at https://github.com/amaranth-lang/amaranth/milestone/4.

In addition, we have made a policy change that will avoid the same issue happening in the future. You can read about it at https://amaranth-lang.org/rfcs/0021-patch-releases.html.

I am now closing this question as no action is planned until the 0.4 release and the 0.4 release will resolve this.

closed 2023-08-22 by whitequark