amaranth-lang/amaranth #727: Lists are no longer accepted as build option overrides

adamgreig, 2022-11-06:
In the 0.3 release, overrides such as `nextpnr_opts` can be specified as a list of strings, which are joined together by the `options` filter in the jinja2 templates. However, #694 adds type checking to `get_override()` that defaults to requiring a string, so a list is rejected before it's passed to `options`.

This ends up being a breaking change from 0.3, but I don't know what behaviour you'd rather have. It seems convenient to keep allowing lists and using the `options` filter, but then `get_override` should probably accept both strings and lists-of-strings by default. Another option is to always require strings and remove the `options` filter altogether.

whitequark, 2022-11-06:
The old behavior was unintentional. The reason lists are rejected is that you cannot pass a list through an environment variable (since those are just strings), and requiring strings means things work the same no matter what.

adamgreig, 2022-11-06:
Fair enough, but why have the custom `options` filter that combines lists into strings if it's unintentional? I was mostly using the overrides as kwargs to the build function, where lists are convenient (but of course I can join them myself).

whitequark, 2023-02-06:
[Discussed](https://libera.irclog.whitequark.org/amaranth-lang/2023-02-06#1675703348-1675704056;) on 2023-02-06 meeting, waiting for @adamgreig to chime in further.

adamgreig, 2023-02-06:
My use for the `*_opts` overrides is all programmatic from my Python software that generates bitstreams for different hardware platforms; various specific platforms override things like SPI flash mode and frequency, while the top-level management sets common things like the nextpnr seed, `-abc9` for yosys, and so on. Internally I have these as lists of arguments so that I can concatenate the platform-specific ones with the general options I apply everywhere.

It's no trouble to join the list together into a single string myself, so I don't mind if that changes especially, but because I'm also not using them from the command line and already have a list of arguments, it was convenient to use them as a list in 0.3.

whitequark, 2023-02-06:
Can I take a look?

closed 2023-02-28 by whitequark

mndza, 2023-03-02:
Commit 9ec7f5b5078903320e7f706d059d5e0178196d51 is breaking previously working `str` overrides such as:
```python
'ecppack_opts': '--compress --freq 38.8'
```
`str` is also `Iterable`, so `kwarg` ends being:
```
- - c o m p r e s s   - - f r e q   3 8 . 8
```

commit 9ec7f5b5 (2023-02-28): build.plat: accept lists as build option overrides.

This was unintentionally removed in 9eb208c3.

Fixes #727.