amaranth-lang/amaranth #728: Toolchain environment override environment variable names now must be upper-case

adamgreig, 2022-11-12:
In 0.3 and before #687, environment variable names for overrides such as `AMARANTH_ENV_Diamond` used the same case as the toolchain name (e.g., `Diamond`). After #687 the use of `tool_env_var()` inadvertently(?) causes all names to require upper-case, e.g. `AMARANTH_ENV_DIAMOND`. The upper-case name doesn't work in 0.3, and the previous mixed-case name no longer works.

whitequark, 2022-11-12:
Thanks! I would be happy to review a PR fixing this issue.

adamgreig, 2022-11-12:
Do you want to revert to the old behaviour or maybe make the environment variable names case-insensitive?

whitequark, 2022-11-12:
> make the environment variable names case-insensitive?

I think this would cause issues where you have both `AMARANTH_ENV_DIAMOND` and `AMARANTH_ENV_Diamond`, unfortunately :(

The impetus for #687 was to "[allow] Platform.toolchain to contain '+' or '-' characters.". Maybe we can fix just that instead?

commit af7c1144 (2022-11-16): Use all-uppercase toolchain_env_var names.

Accepts previous case for backwards compatibility.

Fixes #728.
closed 2022-11-16 by whitequark