amaranth-lang/amaranth #730: lib.data: enum Field width cannot be user-defined

jfng, 2022-11-16:
Repro:
```python3
import enum
from amaranth import *
from amaranth.lib import data

class B(enum.Enum):
    FOO = 0b00
    BAR = 0b01

class A(data.Struct):
    b: B
    c: unsigned(1)

a = A()
print(len(a.b)) # 1
```
The width of an enum Field is currently inferred from ``Shape.cast()``, as the smallest width required to represent all of its members.

In the above example, a user may actually want the ``A.b`` field to have a width of 2 bits. Currently, the enum would need a third member with a value of `0b10` or `0b11`.

whitequark, 2022-11-16:
I'm wondering if there are other cases for defining Enums that are wider when shape-cast than what is inferred from their variants alone, and if so, whether this should actually be fixed in the core language.

jfng, 2022-11-16:
Signals with enum shapes have the same issue, but it can be worked around by using an explicit width, with the enum as a decoder (e.g. ``s = Signal(unsigned(2), decoder=B)``).

Records also have this issue. Their (non-record) fields are signals, so the above workaround can work, with more awkwardness (e.g. ``r = Record([("b", unsigned(2))]); r.b.decoder = B``).

whitequark, 2023-02-06:
[Discussed](https://libera.irclog.whitequark.org/amaranth-lang/2023-02-06#1675706095-1675709437;) on 2023-02-06 weekly meeting. This feature needs an RFC; the discussion can be continued there.

whitequark, 2023-02-21:
I wrote an RFC for this feature: https://github.com/amaranth-lang/rfcs/pull/3

closed 2023-02-27 by whitequark

whitequark, 2023-02-28:
This works now:

```python
from amaranth import *
from amaranth.lib import data, enum

class B(enum.Enum, shape=unsigned(2)):
    FOO = 0b00
    BAR = 0b01

class A(data.Struct):
    b: B
    c: unsigned(1)

a = A()
print(len(a.b)) # 2
```