amaranth-lang/amaranth #735: Repl() expression as Array index work in sim, but not with build

bl0x, 2023-01-13:
Expressions containing Repl() will work fine as Array indices in simulation, but not when building for a platform.
Is this expected?

Amarath is at commit 0a1ba2205064512455e9e9d1ace1b63f0bffffc8

Consider the minimal example:

```python
from amaranth import *
from amaranth.sim import Simulator

import sys

m = Module()

a = Array([Signal(8) for x in range(8)])

b1 = Repl(0, 3) # does not work
b2 = C(0, 3)    # works
b3 = Signal(3)  # works
b4 = Repl(Signal(), 3) # does not work

# choose here between b1...b4
m.d.sync += a[b1].eq(0) 

if len(sys.argv) > 1 and sys.argv[1] == "sim":
    sim = Simulator(m)

    def proc():
        yield

    sim.add_clock(1e-6)
    sim.add_sync_process(proc)

    sim.run_until(2,)
else:
    from amaranth_boards.arty_a7 import *
    platform = ArtyA7_35Platform(toolchain="Symbiflow")
    platform.build(m)
```

Run with
```
python bug.py sim
```
to use the simulator, or with
```
python bug.py
```
to build for the platform.

Simulation will work just fine without complaints, but building yields following errors:

When using `b1`:

```
Traceback (most recent call last):
  File "site-packages/amaranth/back/rtlil.py", line 802, in on_statement
    super().on_statement(stmt)
  File "site-packages/amaranth/hdl/xfrm.py", line 209, in on_statement
    new_stmt = self.on_Assign(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 750, in on_Assign
    self._case.assign(self.lhs_compiler(stmt.lhs), rhs_sigspec)
  File "site-packages/amaranth/hdl/xfrm.py", line 127, in __call__
    return self.on_value(value)
  File "site-packages/amaranth/hdl/xfrm.py", line 112, in on_value
    new_value = self.on_ArrayProxy(value)
  File "site-packages/amaranth/back/rtlil.py", line 430, in on_ArrayProxy
    raise _LegalizeValue(value.index, range(min(max_index, max_elem)), value.src_loc)
amaranth.back.rtlil._LegalizeValue: ((repl (const 1'd0) 3), range(0, 8), ('bugs/repl_index.py', 12))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "site-packages/amaranth/back/rtlil.py", line 368, in expand_to
    assert value not in self.expansions
  File "envs/xc7/lib/python3.7/_collections_abc.py", line 666, in __contains__
    self[key]
  File "site-packages/amaranth/hdl/ast.py", line 1599, in __getitem__
    key = None if key is None else self._map_key(key)
  File "site-packages/amaranth/hdl/ast.py", line 1698, in __init__
    .format(self.value))
TypeError: Object (repl (const 1'd0) 3) cannot be used as a key in value collections

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "bugs/repl_index.py", line 27, in <module>
    platform.build(m)
  File "site-packages/amaranth/build/plat.py", line 107, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "site-packages/amaranth/build/plat.py", line 179, in prepare
    return self.toolchain_prepare(fragment, name, **kwargs)
  File "site-packages/amaranth_boards/arty_a7.py", line 212, in toolchain_prepare
    return super().toolchain_prepare(fragment, name, **overrides, **kwargs)
  File "site-packages/amaranth/build/plat.py", line 324, in toolchain_prepare
    rtlil_text, self._name_map = rtlil.convert_fragment(fragment, name=name)
  File "site-packages/amaranth/back/rtlil.py", line 1042, in convert_fragment
    _convert_fragment(builder, fragment, name_map, hierarchy=(name,))
  File "site-packages/amaranth/back/rtlil.py", line 958, in _convert_fragment
    stmt_compiler(group_stmts)
  File "site-packages/amaranth/hdl/xfrm.py", line 232, in __call__
    return self.on_statement(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 802, in on_statement
    super().on_statement(stmt)
  File "site-packages/amaranth/hdl/xfrm.py", line 220, in on_statement
    new_stmt = self.on_statements(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 820, in on_statements
    self.on_statement(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 814, in on_statement
    with self.state.expand_to(legalize.value, branch_value):
  File "envs/xc7/lib/python3.7/contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "site-packages/amaranth/back/rtlil.py", line 372, in expand_to
    del self.expansions[value]
  File "site-packages/amaranth/hdl/ast.py", line 1607, in __delitem__
    key = None if key is None else self._map_key(key)
  File "site-packages/amaranth/hdl/ast.py", line 1698, in __init__
    .format(self.value))
TypeError: Object (repl (const 1'd0) 3) cannot be used as a key in value collections
```

When using `b4`:

```
Traceback (most recent call last):
  File "site-packages/amaranth/back/rtlil.py", line 802, in on_statement
    super().on_statement(stmt)
  File "site-packages/amaranth/hdl/xfrm.py", line 209, in on_statement
    new_stmt = self.on_Assign(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 750, in on_Assign
    self._case.assign(self.lhs_compiler(stmt.lhs), rhs_sigspec)
  File "site-packages/amaranth/hdl/xfrm.py", line 127, in __call__
    return self.on_value(value)
  File "site-packages/amaranth/hdl/xfrm.py", line 112, in on_value
    new_value = self.on_ArrayProxy(value)
  File "site-packages/amaranth/back/rtlil.py", line 430, in on_ArrayProxy
    raise _LegalizeValue(value.index, range(min(max_index, max_elem)), value.src_loc)
amaranth.back.rtlil._LegalizeValue: ((repl (sig $signal) 3), range(0, 8), ('bugs/repl_index.py', 12))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "site-packages/amaranth/back/rtlil.py", line 368, in expand_to
    assert value not in self.expansions
  File "envs/xc7/lib/python3.7/_collections_abc.py", line 666, in __contains__
    self[key]
  File "site-packages/amaranth/hdl/ast.py", line 1599, in __getitem__
    key = None if key is None else self._map_key(key)
  File "site-packages/amaranth/hdl/ast.py", line 1698, in __init__
    .format(self.value))
TypeError: Object (repl (sig $signal) 3) cannot be used as a key in value collections

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "bugs/repl_index.py", line 27, in <module>
    platform.build(m)
  File "site-packages/amaranth/build/plat.py", line 107, in build
    plan = self.prepare(elaboratable, name, **kwargs)
  File "site-packages/amaranth/build/plat.py", line 179, in prepare
    return self.toolchain_prepare(fragment, name, **kwargs)
  File "site-packages/amaranth_boards/arty_a7.py", line 212, in toolchain_prepare
    return super().toolchain_prepare(fragment, name, **overrides, **kwargs)
  File "site-packages/amaranth/build/plat.py", line 324, in toolchain_prepare
    rtlil_text, self._name_map = rtlil.convert_fragment(fragment, name=name)
  File "site-packages/amaranth/back/rtlil.py", line 1042, in convert_fragment
    _convert_fragment(builder, fragment, name_map, hierarchy=(name,))
  File "site-packages/amaranth/back/rtlil.py", line 958, in _convert_fragment
    stmt_compiler(group_stmts)
  File "site-packages/amaranth/hdl/xfrm.py", line 232, in __call__
    return self.on_statement(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 802, in on_statement
    super().on_statement(stmt)
  File "site-packages/amaranth/hdl/xfrm.py", line 220, in on_statement
    new_stmt = self.on_statements(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 820, in on_statements
    self.on_statement(stmt)
  File "site-packages/amaranth/back/rtlil.py", line 814, in on_statement
    with self.state.expand_to(legalize.value, branch_value):
  File "envs/xc7/lib/python3.7/contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "site-packages/amaranth/back/rtlil.py", line 372, in expand_to
    del self.expansions[value]
  File "site-packages/amaranth/hdl/ast.py", line 1607, in __delitem__
    key = None if key is None else self._map_key(key)
  File "site-packages/amaranth/hdl/ast.py", line 1698, in __init__
    .format(self.value))
TypeError: Object (repl (sig $signal) 3) cannot be used as a key in value collections
```

whitequark, 2023-01-13:
Nope, that's a bug.

bl0x, 2023-01-16:
Oops. Easy to fix, or requires more in-depth knowledge? I could take a look, but haven't read the rtlil code yet.

whitequark, 2023-01-16:
Simple omission in ast.ValueKey. You should be able to fix it.

bl0x, 2023-01-16:
I'll go at it. Thx.

PR #737 (2023-01-16): Add Repl() to ValueKey.
fixes #735
commit 427c82fc (2023-01-16): hdl.ast: handle `Repl` in `ValueKey`.

Fixes #735.
closed 2023-01-16 by whitequark