amaranth-lang/amaranth #742: Regression in Fmax of generated hardware

lekcyjna123, 2023-02-01:
Hi,

we are currently working on out-of-order RiscV processor (I hope that we will be able to publish it's source code in month or so) and as part of that we have implemented benchmark in our CI, to measure some key parameters of generated HW. Recently we have updated amaranth from 0.3 version to newest trunk and after that we observed ~15% degradation of Fmax parameter of generated design (from ~63MHz to ~53MHz).
[image]

We use flag `PYTHONHASHSEED=0` so that our results will be repeatable.

I have done a binary search and it looks like the newest degradation was introduced in commit 
[back.rtlil: avoid sync process emission in RTLIL](https://github.com/amaranth-lang/amaranth/commit/5a4d45b59993206fcc3590cf74ae78c4fba55fe8)
After that commit Fmax dropped from ~57 MHz to ~53 MHz.

I am going to dig deeper to find a second commit, which decreased Fmax from 63MHz to 57MHz. When I find it I will make an update.

whitequark, 2023-02-02:
Unfortunately the information you've provided is insufficient to narrow down the issue or even determine if there is any. The commit 5a4d45b59993206fcc3590cf74ae78c4fba55fe8 is a soundness fix; it is entirely possible that before this commit, there was a functional issue in the netlist emitted by Amaranth that is no longer present afterwards.

lekcyjna123, 2023-02-09:
Thanks for the answer. Are there any additional information which can I provide to narrow down the problem or should I close the issue?

whitequark, 2023-02-09:
Sure. If you provide a minimized example that exhibits the same behavior I will fix it.

lekcyjna123, 2023-03-05:
It is hard to reproduce this issue as minimized example, because of the fact, that even small change in design has big impact on Fmax (up to 6-8%). So I am closing this question.

closed 2023-03-05 by lekcyjna123