amaranth-lang/amaranth #745: WARNING: Generating metadata for package amaranth produced metadata for project name unknown. Fix your #egg=amaranth fragments.

jeremyherbert, 2023-02-10:
Hi,

When trying to install amaranth on Ubuntu 22.04 with python 3.10 using the following command:

```
pip install git+https://github.com/amaranth-lang/amaranth.git#egg=amaranth[builtin-yosys]
```

I receive the following error:

```
Defaulting to user installation because normal site-packages is not writeable
Collecting amaranth[builtin-yosys]
  Cloning https://github.com/amaranth-lang/amaranth.git to /tmp/pip-install-884z3dch/amaranth_dc9f1a8737054446a33d3461926d4c89
  Running command git clone --filter=blob:none --quiet https://github.com/amaranth-lang/amaranth.git /tmp/pip-install-884z3dch/amaranth_dc9f1a8737054446a33d3461926d4c89
  Resolved https://github.com/amaranth-lang/amaranth.git to commit 666ee27fd0fa9fb90109e0e201f878f9ab034c85
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package amaranth produced metadata for project name unknown. Fix your #egg=amaranth fragments.
Discarding git+https://github.com/amaranth-lang/amaranth.git#egg=amaranth: Requested unknown from git+https://github.com/amaranth-lang/amaranth.git#egg=amaranth has inconsistent name: filename has 'amaranth', but metadata has 'unknown'
ERROR: Could not find a version that satisfies the requirement amaranth (unavailable) (from versions: 0, 0.3)
ERROR: No matching distribution found for amaranth (unavailable)
```

After to some googling, I found this issue: https://github.com/pypa/pipenv/issues/5445 - it seems that one must specify `name="amaranth"` as a keyword arg in the `setup.py` file, and that does not appear to be the case in this repo. I think I am also coming across this issue in `amaranth-soc` and `amaranth-boards` when trying to install LUNA as per this issue: https://github.com/greatscottgadgets/luna/issues/194

Could you please have a look at this?

whitequark, 2023-02-10:
The `name` field is specified in [pyproject.toml](https://github.com/amaranth-lang/amaranth/blob/666ee27fd0fa9fb90109e0e201f878f9ab034c85/pyproject.toml#L8).

miek, 2023-02-10:
For what it's worth, I recently had a similar sounding issue when trying to install using `pip install .`, where it would install as a package named "UNKNOWN". Updating pip (to v23.0) fixed it for me, I assume the older version I had wasn't reading the metadata correctly.

whitequark, 2023-02-10:
> The `name` field is specified in [pyproject.toml](https://github.com/amaranth-lang/amaranth/blob/666ee27fd0fa9fb90109e0e201f878f9ab034c85/pyproject.toml#L8).

(I would specify it in `setup.py` as well but I think then `setuptools` would reject it.)

jeremyherbert, 2023-02-11:
Indeed, this is a problem with pip. By default, Ubuntu 22.04 ships with:

```
pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)
```

By upgrading to pip 23, everything works as expected. I'm not sure if you can require a version of pip in the `pyproject.toml` but I guess that might be a fix...

whitequark, 2023-02-11:
> I'm not sure if you can require a version of pip in the `pyproject.toml` but I guess that might be a fix...

I don't think you can, unfortunately. But this is something we can add to the installation instructions.

whitequark, 2023-02-16:
@jeremyherbert @miek I've updated the instructions while fixing #750 -- could either of you check if what's on https://amaranth-lang.org/docs/amaranth/latest/install.html#development-snapshot now works better?

miek, 2023-02-16:
> @miek I've updated the instructions while fixing #750 -- could either of you check if what's on https://amaranth-lang.org/docs/amaranth/latest/install.html#development-snapshot now works better?

Same result I think - the updated instructions still fail for me on an older pip version (I tried 22.0), but work well on pip 23.0.

whitequark, 2023-02-16:
@miek Thank you.

closed 2023-02-16 by whitequark