amaranth-lang/amaranth #746: Building verilog output with synplify pro

ydnatag, 2023-02-10:
Hi!

# Introduction

I'm having some issues trying to build Amaranth verilog outputs with synplify pro (Libero toolchain). The main problem is that synplify does not support initial blocks. Up to the previous `amaranth-yosys` version, it was only an issue for blockrams. Now, Yosys also uses initial block for FSM with power of 2  and more than 8 states.

# Example

```python
from amaranth import *
from amaranth.back.verilog import convert

class Issue(Elaboratable):
    def __init__(self, n, domain):
        self.N = n
        self.domain = domain
        self.out = Signal(range(self.N))

    def elaborate(self, platform):
        m = Module()
        with m.FSM(domain=self.domain) as fsm:
            for i in range(self.N):
                with m.State(f'STATE{i}'):
                    m.d.comb += self.out.eq(i)
                    m.next = f'STATE{(i + 1) % self.N}'
        return m
for n in [5, 8, 10, 16, 20]:
    e = Issue(n, 'sync')
    print(convert(e, 'top', ports=[e.out]))
```

This generates:

* n = 5

```verilog

module top(clk, rst, out);
  reg \$auto$verilog_backend.cc:2083:dump_module$1  = 0;
  input clk;
  wire clk;
  reg [2:0] fsm_state = 3'h0;
  reg [2:0] \fsm_state$next ;
  output [2:0] out;
  reg [2:0] out;
  input rst;
  wire rst;
  always @(posedge clk)
    fsm_state <= \fsm_state$next ;
  always @* begin
    if (\$auto$verilog_backend.cc:2083:dump_module$1 ) begin end
    out = 3'h0;
    casez (fsm_state)
      /* \amaranth.decoding  = "STATE0/0" */
      3'h0:
          out = 3'h0;
      /* \amaranth.decoding  = "STATE1/1" */
      3'h1:
          out = 3'h1;
      /* \amaranth.decoding  = "STATE2/2" */
      3'h2:
          out = 3'h2;
      /* \amaranth.decoding  = "STATE3/3" */
      3'h3:
          out = 3'h3;
      /* \amaranth.decoding  = "STATE4/4" */
      3'h4:
          out = 3'h4;
    endcase
  end
[... 241 lines cut ...]
    if (\$auto$verilog_backend.cc:2083:dump_module$9 ) begin end
    \fsm_state$next  = \$auto$proc_rom.cc:149:do_switch$1 ;
    casez (rst)
      1'h1:
          \fsm_state$next  = 4'h0;
    endcase
  end
endmodule
```

# Synplify pro warning

```
@W: Within an initial block, only Verilog force statements and memory $readmemh/$readmemb initialization statements are recognized, and all other content is ignored. Simulation mismatch may occur
```

# Current workaround

Currently, I'm using regular expressions to extract all the initial blocks and change the syntax to something synplify supports using `$readmemh`.

# Question

I know that Libero is not a supported toolchain in amaranth, so it is not a real issue. But it would be great if there is a way to avoid initial block emission. If it is too difficult to fix this or it is not in the roadmap, can any of you suggest me a better workaround?

Thanks

whitequark, 2023-02-10:
While Libero is not specifically declared as supported (since there's no way at the moment to ensure this support keeps functioning), the Verilog emitted by Amaranth is supposed to have a wide compatibility. In particular it should obviously work with Synplify, which is probably the most common Verilog frontend vendor.

I'll investigate this.

ydnatag, 2023-02-10:
Great, let me know if I can help.

whitequark, 2023-02-10:
Try this patch:
```patch
diff --git a/amaranth/back/verilog.py b/amaranth/back/verilog.py
index 738080c..dcddb54 100644
--- a/amaranth/back/verilog.py
+++ b/amaranth/back/verilog.py
@@ -15,7 +15,7 @@ def _convert_rtlil_text(rtlil_text, *, strip_internal_attrs=False, write_verilog
 
     script = []
     script.append("read_ilang <<rtlil\n{}\nrtlil".format(rtlil_text))
-    script.append("proc -nomux")
+    script.append("proc -nomux -norom")
     script.append("memory_collect")
 
     if strip_internal_attrs:
```

ydnatag, 2023-02-14:
The patch works for new yosys version and the verilog syntax is always "casez". Thanks!!!

The first time I tried to test this patch, I forgot to add `AMARANTH_USE_YOSYS=builtin` env var and  I got the following error:

```
amaranth._toolchain.yosys.YosysError: ERROR: Command syntax error: Unknown option or option in arguments.
> proc -nomux -norom
```

It looks like that `norom` is not supported for previous versions of yosys.

---

Just to know, Do you have something in mind to change the memory initialization syntax? I hate the solution I have now  using regex.

whitequark, 2023-02-15:
> The patch works for new yosys version and the verilog syntax is always "casez". Thanks!!!

Thanks. I'll add a workaround that functions with earlier versions of Yosys as well in a bit.

> Just to know, Do you have something in mind to change the memory initialization syntax? I hate the solution I have now using regex.

Yes, this has been a problem for a while. It is probably going to be done in medium term.

commit 0b7adcbd (2023-02-28): back.verilog: pass `-norom` to `-proc` if available.

Fixes #746.
closed 2023-02-28 by whitequark