amaranth-lang/amaranth #764: GTKW files with signal groups

tilk, 2023-03-09:
For debugging complex circuits, it's useful to have the signal traces grouped according to their function. The popular _gtkwave_ tool supports this, as does the `GTKWSave` class from the `vcd` package used by Amaranth to export VCD files. All that remains is to connect this together.

I have [implemented an extension](https://github.com/kuznia-rdzeni/coreblocks/blob/master/test/gtkw_extension.py) to Amaranth's `_VCDWriter` which does just that. The idea is that the `traces` argument can be passed other things than lists of signals. My implementation currently accepts the following as traces:

* Signals, which work as before.
* Iterables (e.g. lists) of traces, which shows them in _gtkwave_ one after the other, like before.
* Mappings (e.g. dicts) of traces, which create a new group of traces for each item in the mapping,
* Amaranth Records, which group the fields of the record.

There is no support in the current implementation for the recently added `lib.data` classes from RFC 1, but it can be easily added.

Are you interested in merging this functionality to Amaranth?

whitequark, 2023-03-09:
Let's discuss improvements to VCD and GTKW writing functionality after the next release.

whitequark, 2024-03-22:
I'm interested in expanding `traces=` handling with the following behavior:
- `dict`: creates a group per dict key.
- `list` or `tuple`: expands into individual elements recursively.
- `data.View`: expands into a group with individual parts, unless a single field, in which case the group is omitted.
- `MemoryData`: expands into all rows of the memory (requires #1221).
- `wiring` interface objects: expands into a group with individual signals recursively.

Note that I've specified `dict` and `list` above because I consider the syntax of tracing to be a mini-DSL where I ascribe new meaning to, specifically, the Python `{}` and `[]`/`()` syntax. This allows flexibility for later expansion with other objects, which may also be e.g. iterables.

Is this something you're interested in implementing? I'd be happy to merge that in 0.5, but otherwise I may have to bump it to 0.6.

tilk, 2024-03-22:
I could port the implementation of this feature from Coreblocks, adapting the behavior to stated requirements.

whitequark, 2024-03-22:
Sounds good to me.

tilk, 2024-03-25:
It looks like there is a regression on #790: I can't get view field traces on main. I will try to do a bisect later.

tilk, 2024-03-25:
I bisected the regression to commit 2bf1b4dafc3a5860e532ae8dac31859c3f9c7089.

wanda-phi, 2024-03-25:
Yeah, I see the problem, I'm going to do a quick fix soon

wanda-phi, 2024-03-25:
Fixed in #1237.

PR #1245 (2024-05-22): Emit signal groups in GTKW files
Fixes #764.

Things to look at in the review:
* Group names for views and interfaces. The current code uses "view" and "interface". Maybe the name should reference a layout/signature in some way?
* Treating single-field view as in https://github.com/amaranth-lang/amaranth/issues/764#issuecomment-2014220740 is not implemented. It could be implemented by adding a special case.
* I've added tests that check if the code doesn't raise an exception, the resulting gtkw files are not checked. I'm not sure how to best check them; I could compare them with a reference, b [... cut]
closed 2024-05-22 by whitequark