amaranth-lang/amaranth #766: pip install broken

daniestevez, 2023-03-12:
Commit 732d62e has broken installation of amaranth via pip (at least in an Ubuntu 20.04 Docker image I'm using).

With the previous commit:
```
root@maia-sdr-devel:/home/user/amaranth# git checkout cd7dfee
[...]
root@maia-sdr-devel:/home/user/amaranth# pip install .
Processing /home/user/amaranth
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting pyvcd<0.4,>=0.2.2
  Downloading pyvcd-0.3.0-py2.py3-none-any.whl (23 kB)
Collecting Jinja2~=3.0
  Downloading Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 KB 1.5 MB/s eta 0:00:00
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (25 kB)
Building wheels for collected packages: amaranth
  Building wheel for amaranth (pyproject.toml) ... done
  Created wheel for amaranth: filename=amaranth-0.4.dev53+gcd7dfee-py3-none-any.whl size=148301 sha256=41b308333f4ba253a58a864b133d491643deb65f2a760f0d7a11af1d11ef3399
  Stored in directory: /tmp/pip-ephem-wheel-cache-oq_m2o2f/wheels/2e/b7/37/2b538126c2fdbcdf5cf642269f3ec577d7d0f506368e00d48f
Successfully built amaranth
Installing collected packages: pyvcd, MarkupSafe, Jinja2, amaranth
Successfully installed Jinja2-3.1.2 MarkupSafe-2.1.2 amaranth-0.4.dev53+gcd7dfee pyvcd-0.3.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```
at this point I can run `import amaranth` in a Python interpreter and it works.

With commit 732d62e,
```
root@maia-sdr-devel:/home/user/amaranth# git checkout 732d62e
[...]
root@maia-sdr-devel:/home/user/amaranth# pip install .
Processing /home/user/amaranth
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: UNKNOWN
  Building wheel for UNKNOWN (pyproject.toml) ... done
  Created wheel for UNKNOWN: filename=UNKNOWN-0.4.dev54+g732d62e-py3-none-any.whl size=2142 sha256=fac2e3e1f913df3ed631792193c68ec8f96b51669c53997d247f77ee1764a634
  Stored in directory: /tmp/pip-ephem-wheel-cache-v985v5fu/wheels/2e/b7/37/2b538126c2fdbcdf5cf642269f3ec577d7d0f506368e00d48f
Successfully built UNKNOWN
Installing collected packages: UNKNOWN
Successfully installed UNKNOWN-0.4.dev54+g732d62e
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```
I cannot `import amaranth` in a Python interpreter, and the install directory looks like this:
```
root@maia-sdr-devel:/home/user/amaranth# ls /usr/local/lib/python3.10/dist-packages/UNKNOWN-0.4.dev54+g732d62e.dist-info/
INSTALLER  LICENSE.txt  METADATA  RECORD  REQUESTED  WHEEL  direct_url.json  top_level.txt
```
It seems that none of the `.py` files get installed anywhere in `/usr`.

whitequark, 2023-03-12:
Please run `pip install -U pip` first and recheck.

daniestevez, 2023-03-12:
Thanks for very quick reply! Upgrading pip works. I can now install correctly https://github.com/amaranth-lang/amaranth/commit/732d62eb2412e38f6a0ccd4a6efc59a5c73898b7 as well as the current `main`. This issue can be closed, I think.

whitequark, 2023-03-12:
This is [documented](https://amaranth-lang.org/docs/amaranth/latest/install.html#installing-prerequisites); I know most people don't re-read this section but unfortunately pip has issues like this sometimes.

closed 2023-03-12 by whitequark

daniestevez, 2023-03-12:
My bad for not reading the manual. I promise I glanced at that page to check if I was doing something wrong, but I didn't look at that section too attentively.

whitequark, 2023-03-12:
I could probably make pip print a warning message during installation...