amaranth-lang/amaranth #769: Possible to define operators for aggregate types?

matthuszagh, 2023-05-01:
Is it possible to define operators for aggregate data types? For example, consider a simple aggregate data type representing a complex number:

```python
class Complex(data.View):
    re: Signal
    im: Signal

    def __init__(self, width: int):
        super().__init__(
            data.StructLayout({"re": unsigned(width), "im": unsigned(width)}),
            None,
        )
```

To add one complex aggregate to another we could do something like:

```python
a = Complex(16)
b = Complex(16)
c = Complex(16)
c.re.eq(a.re + b.re)
c.im.eq(a.im + b.im)
```

Or, to add a constant to a complex number:

```python
a = Complex(16)
b = Complex(16)
a.re.eq(b.re + 15)
a.im.eq(b.im)
```

However, it might be nice to simply write

```
c.eq(a + b)
```

or,

```
a.eq(b + 15)
```

Is it possible to define `__add__` in a way that makes these shorthands possible?

whitequark, 2023-05-02:
Yep! You can define `__add__` that checks for another `Complex` instance and otherwise casts to a `Value`.

This (and `Fixed`) might actually end up in the stdlib sooner or later.

matthuszagh, 2023-05-02:
Excellent, thanks!

Based on that, the following seems to work (but correct me if I've misunderstood, as I'm pretty new to Amaranth):

```python
    def __add__(self, other):
        if isinstance(other, Complex):
            return Cat(self.re + other.re, self.im + other.im)
        else:
            return Cat(self.re + other, self.im)
```

whitequark, 2023-05-02:
> Based on that, the following seems to work (but correct me if I've misunderstood, as I'm pretty new to Amaranth):

You might want to return a `Complex` instance from here?

matthuszagh, 2023-05-02:
Ok, attempt 2:

```python
class Complex(data.View):
    re: Signal
    im: Signal

    def __init__(self, value=None, *, rwidth: int, iwidth: int):
        super().__init__(
            data.StructLayout(
                {"re": unsigned(rwidth), "im": unsigned(iwidth)}
            ),
            value,
        )

    def __add__(self, other):
        if isinstance(other, Complex):
            re_val = self.re + other.re
            im_val = self.im + other.im
        else:
            re_val = self.re + other
            im_val = self.im

        return Complex(
            Cat(re_val, im_val), rwidth=len(re_val), iwidth=len(im_val)
        )

    __radd__ = __add__

    def eq(self, other):
        return Assign(
            self, Cat(other.re[0 : len(self.re)], other.im[0 : len(self.im)])
        )
```

I also defined `eq` so that any MSB truncation is equally shared between the `re` and `im` fields. Otherwise I think this would cast to `Value` and truncate the imaginary part first.

whitequark, 2023-05-02:
Your `__eq__` method needs a type check or you'll get an AttributeError if you try to assign a `Value` to it.

matthuszagh, 2023-05-02:
Ah right of course. So something like this:

```python
    def eq(self, other):
        if isinstance(other, Complex):
            val = Cat(other.re[0 : len(self.re)], other.im[0 : len(self.im)])
        else:
            val = other

        return Assign(self, val)
```

whitequark, 2023-05-02:
Yes. I think slightly more work needs to be done to handle every possible case (I don't quite remember every detail offhand) but that's the gist of it.

matthuszagh, 2023-05-02:
> Yes. I think slightly more work needs to be done to handle every possible case (I don't quite remember every detail offhand) but that's the gist of it.

Cool no worries. I'll post here if I run into anything.

Also, in case anyone else looks at this and it isn't obvious: my definition of `Complex` isn't great. It would often be desirable to use signed values for `re` and `im`, and it would be nice to be able to provide just one width for both fields in many cases. But, this should be easy to achieve by modifying `__init__`; this was just meant to be a basic example.

whitequark, 2023-05-02:
I think that it would be a lot of value if someone wrote an RFC for a generic (over underlying numeric type) `Complex` type for inclusion in the standard library.

`Fixed` type as well!

matthuszagh, 2023-05-02:
> I think that it would be a lot of value if someone wrote an RFC for a generic (over underlying numeric type) `Complex` type for inclusion in the standard library.

I can work on that.

> `Fixed` type as well!

Yeah that would be really useful too. Let me give that some thought.

whitequark, 2023-11-25:
I believe the question has been answered, so I'll close the issue.

Note that @zyp has been working on a fixed point number RFC for inclusion in the standard library, but complex numbers are still up for grabs.

closed 2023-11-25 by whitequark