amaranth-lang/amaranth #786: Introspection of View?

tilk, 2023-05-20:
Is there any particular reason why it's not possible to iterate over the fields of a `View`, or at least access the underlying `Layout` of the view? I'm in the process of assessing the possibility of migration from records to data structures, and it's a bummer that in many places instead of one entity (the record), I need to pass around two (the signal/view and its layout).

whitequark, 2023-05-20:
You can. `Layout.of(view)`

tilk, 2023-05-20:
I don't know why I couldn't discover this.

closed 2023-05-20 by tilk

bieganski, 2023-05-27:
`Layout.of(x)` assumes `x` to be instantiated `View`, it doesn't work when passing type.

It leads to boilerplate in my use case, when using "Ergonomic layout definition":

let's consider such example:

```py
class A(data.View):
   a : unsigned(123)
   b : unsigned(456)
   c : unsigned(789)
   
layout = Layout.of(A(Signal(123+456+789))
```  

what is the "proper" way of accessing layout in such scenario?

bieganski, 2023-05-27:
@whitequark , I will leverage the fact that i have your attention and ask: I understand that lack of default `target` (that would allocate a `Signal`, not yet referenced by any other) argument of a `View` is intentional? 

In the RFC0001, in the example below, it looked to me that `data.Struct` instantiation will allocate a Signal, as we can instantiate `SomeVariant` without any params passed to constructor.

```py
class SomeVariant(data.Struct):
    class Value(data.Union):
        one_signed: signed(2)
        two_unsigned: data.ArrayLayout(unsigned(1), 2)

    kind: Kind
    value: Value

# this class can be used in the same way as a `data.View` without needing to specify the layout:
view2 = SomeVariant()
m.d.comb += [
    view2.kind.eq(Kind.ONE_SIGNED),
    view2.value.eq(view.value)
]
```

whitequark, 2023-05-27:
> Layout.of(x) assumes x to be instantiated View, it doesn't work when passing type.

This is a bug, and it has the same underlying cause as https://github.com/amaranth-lang/amaranth/issues/794.

> I understand that lack of default `target` (that would allocate a `Signal`, not yet referenced by any other) argument of a `View` is intentional?

It used to be that way, but it has been updated by [RFC 15](https://amaranth-lang.org/rfcs/0015-lifting-shape-castables.html). See that RFC for details.

hannesha, 2023-05-29:
I encountered this before and came up with the following workaround:
Wrapping `Layout.of` with `Layout.cast` produces the expected behavior.

The following example demonstrates introspection of any `data.View` class by printing every field:
```python
def print_view(view: data.View):
	layout = data.Layout.cast(data.Layout.of(view))

	for name, shape in layout:
		print(name, shape)
		field_value = view[name]
		if isinstance(field_value, data.View):
			s = yield field_value.as_value()
		else:
			s = yield field_value
		print(name, f'{s:X}')
```

whitequark, 2023-08-23:
Note that `data.Layout.of(view)` is going to be imminently removed and replaced with `view.shape()`. This also fixes `Signal.like(view)`.